RPi-Distro/pi-gen #305: First docker build fails to configure base packages

matt-duke, 2019-07-05:
Pulling from master and running build-docker.sh fails to bootstrap stage0. A second run using CONTINUE=1 ./build-docker.sh is successful.
The issue involves raspbian-archive-keychain. The bootstrap log (attached) shows:
`Setting up gnupg (2.2.12-1+rpi1) ...
Setting up raspbian-archive-keyring (20120528.2) ...
Warning: apt-key should not be used in scripts (called from postinst maintainerscript of the package raspbian-archive-keyring)
gpg: can't connect to the agent: IPC connect call failed
dpkg: error processing package raspbian-archive-keyring (--configure):
 installed raspbian-archive-keyring package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of apt:
 apt depends on raspbian-archive-keyring; however:
  Package raspbian-archive-keyring is not configured yet.

dpkg: error processing package apt (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of apt-utils:
 apt-utils depends on apt (= 1.8.2); however:
  Package apt is not configured yet.

dpkg: error processing package apt-utils (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of tasksel:
 tasksel depends on apt; however:
  Package apt is not configured yet.

dpkg: error processing package tasksel (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of tasksel-data:
 tasksel-data depends on tasksel (= 3.53); however:
  Package tasksel is not configured yet.

dpkg: error processing package tasksel-data (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-5+rpi1) ...
Errors were encountered while processing:
 raspbian-archive-keyring
 apt
 apt-utils
 tasksel
 tasksel-data
Setting up raspbian-archive-keyring (20120528.2) ...
Warning: apt-key should not be used in scripts (called from postinst maintainerscript of the package raspbian-archive-keyring)
gpg: can't connect to the agent: IPC connect call failed
dpkg: error processing package raspbian-archive-keyring (--configure):
 installed raspbian-archive-keyring package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of apt:
 apt depends on raspbian-archive-keyring; however:
  Package raspbian-archive-keyring is not configured yet.`

I'm stumped. Any advice? This was running on Ubuntu 18.04 32-bit VM.

[debootstrap.log](https://github.com/RPi-Distro/pi-gen/files/3364202/debootstrap.log)

hhromic, 2019-07-06:
I just build an image using `build-docker.sh` last night without any issue.
Can you provide the full command you are using for the build? Can you provide more info about the env you are using for the build? Docker version, Linux version, etc.

matt-duke, 2019-07-06:
My mistake, the VM is 64-bit, which it needs to be in order to install Docker.
The VM is running Ubuntu installed from the minimal iso on a PROXMOX host. 
docker -v: "Docker version 18.09.7, build 2d0083d"
uname -r: "4.15.0-54-generic"
Version from /etc/os-release: "18.04.2 LTS (Bionic Beaver)"

My config file is in the pi-gen folder, so the command I'm using is just `sudo ./build-docker.sh`

matt-duke, 2019-07-06:
My mistake, the VM is 64-bit, which it needs to be in order to install Docker.
The VM is running Ubuntu installed from the minimal iso on a PROXMOX host. 
docker -v: "Docker version 18.09.7, build 2d0083d"
uname -r: "4.15.0-54-generic"
Version from /etc/os-release: "18.04.2 LTS (Bionic Beaver)"

matt-duke, 2019-07-10:
I've been unable to repeat this issue in other virtual machines or systems running a nearly identical setup. I will close this issue based on the assumption its related to my configuration. If I ever figure out the root cause I'll update it here.

closed 2019-07-10 by matt-duke