RPi-Distro/pi-gen #309: buster InRelease' changed its 'Suite' value from 'testing' to 'stable'

bzuelly, 2019-07-08:
I'm building on a physical box running 32-bit ubuntu server.

Its been working all day until just a little bit ago.  It now stops shortly after starting and  says **"buster InRelease' changed its 'Suite' value from 'testing' to 'stable'"**

Anyone else seeing this error?

hhromic, 2019-07-08:
I just built a test image without any issues using a Docker build based on `i386/debian:buster`.
While is true this is not the same setup as yours, it still means `pi-gen` itself is probably okay,

Debian Buster this weekend was promoted from `testing` to `stable`, perhaps you should clean your work directories (sysroots) and start the build 100% from scratch?

hhromic, 2019-07-08:
I just got the mentioned error on a **running** Buster image that I previously built (before this weekend) while doing `apt-get update`:
```
# apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```

This StackExchange answer provides an easy fix (and explanation): https://superuser.com/a/1457007
> What happens here, is that the local apt database remembered your package-sources "release-information" to be "busterAsTesting", and when updating now "busterAsStable" is returned.

This further makes me think that you just need to start your builds from scratch so you invalidate your old cache containing `testing` as suite.

In the running image, after doing `apt-get update --allow-releaseinfo-change` once, the cache gets updated and the error goes away for good.

bzuelly, 2019-07-08:
whats the best way to clean my work directories?  purge "work/"?

hhromic, 2019-07-10:
@bzuelly yes probably cleaning the `work` directory should be enough.

XECDesign, 2019-07-23:
ok to close?

bzuelly, 2019-07-23:
Yes thanks for your help

closed 2019-07-23 by XECDesign