RPi-Distro/pi-gen #312: Question: Recommended way of running qemu images

joaocenoura, 2019-07-13:
Hi guys, I'm a n00b qemu + started recently using this project. I've freshly built a rpi image with `QEMU=1` and have the `*-qemu.img` in place. My goal was to run these locally (ssh would be enough) without having to burning it to an SD and load it in the RPI. I've searched a bit on how to run these but saw few different ways of running a regular `*.img` file.
So I'm wondering if there are recommended commands to run `*-qemu.img` in this project or other better way of testing the images locally.

Thanks

XECDesign, 2019-07-15:
That was added by @jacen92 . Perhaps they may be able to answer.

jacen92, 2019-07-15:
Hi,

You could follow this: https://github.com/jacen92/pi-gen/wiki/QEMU
Not tested since jessie version.

As soon as I have the time I will test with the Buster version.

XECDesign, 2019-09-25:
No activity

closed 2019-09-25 by XECDesign