RPi-Distro/pi-gen #316: built images (stage4) has grey desktop without any icons

johwick12, 2019-07-25:
I built an image from master (upto stage4). 
There were no issues during build however the built image booted up to a grey desktop. 
Most of the icons/images are missing.  
Any idea what might be causing this issue?

I am using Ubuntu 16.04 64 bit to build the image.

rvoitenko, 2019-07-26:
Same for me, but using docker build.

XECDesign, 2019-07-26:
Same issue as https://github.com/RPi-Distro/pi-gen/issues/271. You could try using qemu-user-static:i386 from multiarch support and doing a new clean build.

mirceacaprioru, 2019-07-30:
This is happening because of libgdk-pixbuf.

For some reason it is not included in the install-packages and if you add it manually it should work but it does. 

What i managed to do is install it manually from source.

johwick12, 2019-08-06:
Built it on a ubuntu 16.04 32 bit but that didnt help. libgdk-pixbuf is already installed so thats not the issue for me.
I will try using using qemu-user-static:i386 as suggested by @XECDesign.

mikehash, 2019-08-07:
I've been having the same issue with the buster release, a workaround is to run "/usr/lib/arm-linux-gnueabihf/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders --update-cache" after first boot, but it's not practical for distribution.

I just tried building the image using a 32bit VM and still getting the same issue.
Trying to build now with an ARM ec2 instance.

XECDesign, 2019-08-07:
The missing icons are one of many issues. Just fixing the icons after the fact is not a solution or a workaround, since the resulting system is still broken in other ways.

I think an ARM VPS would be aarch64 only running on cavium thunderx or similar, so it wouldn't work.

Since I've moved to building on ARM, I haven't tested some of the proposed fixes yet like qemu-user-static:i386, but others have reported it works.

mikehash, 2019-08-07:
> The missing icons are one of many issues. Just fixing the icons after the fact is not a solution or a workaround, since the resulting system is still broken in other ways.
>
> I think an ARM VPS would [...]

Thank you for the quick reply, I completely agree and that's why I haven't really used the Buster release for production, still using stretch here.
Which ARM server/vps are you using for building?

XECDesign, 2019-08-07:
I couldn't find an ARMv6/7/8 VPS provider. We ended up getting one of these https://www.gigabyte.com/Server-Motherboard/MP30-AR0-rev-11

However, the performance is not much better than a pi 4, so if we were doing it again, a pi 4 would be more than plenty. There's currently another issue causing problems even when building on an ARM machine - https://github.com/RPi-Distro/pi-gen/issues/294

That one is easy to work around.

It's all a bit messy with the Buster release, unfortunately.

mikehash, 2019-08-07:
Sounds good, and I do have a pi4 here which I've been using for testing the buster images.Time to build on it! Thank you :)

cshoredaniel, 2020-03-12:
FYI it seems upgrades via apt cause this same issue for some folks, so it may be a Raspbian issue: https://www.raspberrypi.org/forums/viewtopic.php?p=1276705#p1496865 

As you might have guessed I ran into this with recent build, and was hoping for a quick fix -- no such luck it appears.

cshoredaniel, 2020-03-12:
NVM: I realized I lost the i386 on my Dockerfile, and that the forums posts are rather old (so probably before you started building on ARM itself).

XECDesign, 2020-03-12:
We do enough testing to make sure images with this problem never got shipped. Although the symptoms are similar, it's a different issue.

cshoredaniel, 2020-03-12:
Sorry, was so embarrassed about the disappearing FROM i386/... that I didn't think before I wrote.  It's kind of obvious issue, and for sure you wouldn't ship like that.  Interesting that this particular component seems to be prone to breakage though.

dantrevino, 2020-04-27:
Is there a resolution for this that would enable building properly on x86_64 systems?  Installing qemu-user-static:i386 did not work for me.  The fix mentioned [above](https://github.com/RPi-Distro/pi-gen/issues/316#issuecomment-519179303) works, but after the fact.

XECDesign, 2020-04-27:
Not sure why it doesn't work for you, but I've been building on x64 for the past few weeks using qemu-arm-static from the i386 build in https://packages.ubuntu.com/eoan/qemu-user-static. No issues so far.

dantrevino, 2020-05-14:
Thanks @XECDesign.  I am using Ubuntu 18.04 and that was not working for me.  Updating qemu-user-static:i386 to the eoan version helped!