RPi-Distro/pi-gen #317: Installing ca-certificates-java errors

wb9688, 2019-07-26:
When I try to build it, at some point it exits with the following error:

```
Processing triggers for ca-certificates (20190110) ...
Updating certificates in /etc/ssl/certs...
qemu: Unsupported syscall: 382
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

qemu: Unsupported syscall: 345
Error occurred during initialization of VM
getcpu(2) system call not supported by kernel
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
done.
Processing triggers for shared-mime-info (1.10-1) ...
Processing triggers for sgml-base (1.29) ...
Setting up x11proto-dev (2018.4-4) ...
Processing triggers for fontconfig (2.13.1-2) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Setting up libxau-dev:armhf (1:1.0.8-1+b2) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Setting up libice-dev:armhf (2:1.0.9-2) ...
Setting up libgconf-2-4:armhf (3.2.6-5) ...
Processing triggers for gnome-menus (3.31.4-3) ...
Setting up libsm-dev:armhf (2:1.2.3-1) ...
Processing triggers for libglib2.0-0:armhf (2.58.3-2) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Setting up scratch2 (0.23) ...
Setting up libxdmcp-dev:armhf (1:1.1.2-3) ...
Setting up x11proto-core-dev (2018.4-4) ...
Setting up gconf-service (3.2.6-5) ...
Setting up libxcb1-dev:armhf (1.13.1-2) ...
Setting up libx11-dev:armhf (2:1.6.7-1) ...
Setting up libxt-dev:armhf (1:1.1.5-1) ...
Errors were encountered while processing:
 ca-certificates-java
E: Sub-process /usr/bin/dpkg returned an error code (1)
Command exited with non-zero status 100
```

I also see a lot of other warnings that from qemu about an unsupported syscall. I'm running `build.sh` (so no Docker) from commit 5436273ec728c8369dab9c08f2739805f20510f7 on Ubuntu 18.04.

hhromic, 2019-07-26:
This is a know issue. See near the end of the READMe file. Refer to: https://github.com/RPi-Distro/pi-gen/issues/271
Try using the `qemu-user-static:i386` package from multiarch instead of the 64 bits version.

closed 2019-08-08 by wb9688

felicianotech, 2019-08-22:
I'm having this same issue and installing `qemu-user-static:i386` didn't fix it.

XECDesign, 2019-08-22:
Did you do a full clean build from scratch after installing that?