RPi-Distro/pi-gen #319: initial filesystem resize with size option

fourstone, 2019-08-02:
Hi,

there are a whole number of reasons one might not want the whole disk filled with a single root partition, let me name a few:
- huge SD-cards, e.g. beyond 16GB
- booting directly off an USB-attached SSD, easily 100GB or more
- worrying on sector wear and wanting to leave some space unpartitioned for wear leveling/overprovisioning

Suggestion: Additional optional parameter for resize on first boot. can be entered in cmdline.txt
value should be in GB giving the size to fill. 
16 -> fill the first 16GB of the drive

I think that would solve most of the problems users have and want to create custom partitions.
Not extending root filesystem seems to be a bad option as you cannot install any software with a full root filesystem.

thanks, Martin

XECDesign, 2019-08-02:
Not right now, sorry. I would recommend taking the init= param out of cmdline and setting up the partitions as you see fit for your use case instead.

closed 2019-08-02 by XECDesign

fourstone, 2019-08-02:
good style to just close this as this obviously is an issue as there is no easy fix if you want to automate