RPi-Distro/pi-gen #320: Building pi-gen via docker gives following errors and fails

PhistterbutInc, 2019-08-04:
I'm using a VMware install of Ubuntu 18.04lts to build via docker using command 'sudo ./build-docker.sh -c myconf' where my conf contains 'IMAGE_NAME='Rasbian' and 'ENABLE_SSH=1'.

the error also happens when using 'sudo CONTINUE=1 ./build-docker.sh -c myconf'

I also had package errors previously and added the line 'sudo apt-get update' to line 34 of 'build.sh' and that fixed them.

the error is as follows: (I've truncated it a bit)

`Setting up libfastinfoset-java (1.2.12-3) ...
Setting up libxml-java (1.1.6.dfsg-3) ...
Setting up libplexus-io-java (3.1.1-1) ...
Setting up libsisu-inject-java (0.3.3-1) ...
Setting up libsisu-plexus-java (0.3.3-3) ...
Setting up libplexus-archiver-java (3.6.0-2) ...
Setting up libmaven3-core-java (3.6.0-1) ...
Setting up libmaven-shared-io-java (3.0.0-3) ...
Setting up libmaven-file-management-java (3.0.0-1) ...
Setting up libistack-commons-java (3.0.6-3) ...
Setting up libreoffice-style-tango (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-style-colibre (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-common (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libcodemodel-java (2.6+jaxb2.3.0.1-8) ...
Setting up libtxw2-java (2.3.0.1-8) ...
Setting up libreoffice-core (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-math (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-gtk3 (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libjaxb-java (2.3.0.1-8) ...
Setting up libreoffice-sdbc-postgresql (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-draw (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-java-common (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-base-drivers (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-wiki-publisher (1.2.0+LibO6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-avmedia-backend-gstreamer (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-help-common (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-l10n-en-gb (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libapache-poi-java (4.0.1-1) ...
Setting up libreoffice-sdbc-firebird (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-impress (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-base-core (1:6.1.5-3+rpi1+deb10u2) ...
Setting up python3-uno (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-script-provider-bsh (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-script-provider-js (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-calc (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-base (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-help-en-gb (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-sdbc-hsqldb (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-writer (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-script-provider-python (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-nlpsolver (0.9+LibO6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-report-builder-bin (1:6.1.5-3+rpi1+deb10u2) ...
Setting up liblayout-java (0.2.10-3) ...
Setting up libreoffice (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libreoffice-librelogo (1:6.1.5-3+rpi1+deb10u2) ...
Setting up libpentaho-reporting-flow-engine-java (0.9.4-5) ...
Setting up libreoffice-pi (1:6.1.5-3+rpi1) ...
Setting up libreoffice-report-builder (1:6.1.5-3+rpi1+deb10u2) ...
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-5+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for gnome-icon-theme (3.12.0-3) ...
Processing triggers for shared-mime-info (1.10-1) ...
Processing triggers for fontconfig (2.13.1-2) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
[06:48:38] End /pi-gen/stage5/00-install-libreoffice/00-packages
[06:48:38] End /pi-gen/stage5/00-install-libreoffice
[06:48:38] End /pi-gen/stage5
[06:48:38] Begin /pi-gen/export-image
[06:48:38] Begin /pi-gen/export-image/prerun.sh

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x4ace0865.

Command (m for help): Created a new DOS disklabel with disk identifier 0xc2bc26e3.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): 
Using default response p.
Partition number (1-4, default 1): First sector (2048-4300799, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (8192-4300799, default 4300799): 
Created a new partition 1 of type 'Linux' and of size 256 MiB.

Command (m for help): Disk /pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img: 2.1 GiB, 2202009600 bytes, 4300800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc2bc26e3

Device                                                                    Boot Start    End Sectors  Size Id Type
/pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img1       8192 532480  524289  256M 83 Linux

Command (m for help): Selected partition 1
Hex code (type L to list all codes): Changed type of partition 'Linux' to 'W95 FAT32 (LBA)'.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): 
Using default response p.
Partition number (2-4, default 2): First sector (2048-4300799, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (540672-4300799, default 4300799): 

Created a new partition 2 of type 'Linux' and of size 1.8 GiB.

Command (m for help): 
Command (m for help): 
Disk /pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img: 2.1 GiB, 2202009600 bytes, 4300800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc2bc26e3

Device                                                                    Boot  Start     End Sectors  Size Id Type
/pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img1        8192  532480  524289  256M  c W95 FAT32 (LBA)
/pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img2      540672 4300799 3760128  1.8G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

losetup: /pi-gen/work/2019-08-04-Rasbian/export-image/2019-08-04-Rasbian-lite.img: failed to set up loop device: No such file or directory

real	73m26.329s
user	0m0.552s
sys	0m1.696s
`

PhistterbutInc, 2019-08-04:
Deleting the folders under stage5 named EXPORT* also didn't work.

simplextech, 2019-08-05:
I'm having the exact same issue.  It continues to fail with the same loopback setup error.

PhistterbutInc, 2019-08-05:
Seems to me that there is no folder in the location created to place the files into. I go into /pi-gen and nothing, no work folder. I'm not terribly familiar with the scripts but are we sure these scripts know to make the files?

I tried to make a 'work' folder inside /pi-gen but that didn't work either, same error.

simplextech, 2019-08-06:
Give this a try.  Just did this and finally a build is finished.

Edit:  export-image/prerun.sh

`sleep 5

BOOT_DEV=$(losetup --show -f -o "${BOOT_OFFSET}" --sizelimit "${BOOT_LENGTH}" "${IMG_FILE}")
ROOT_DEV=$(losetup --show -f -o "${ROOT_OFFSET}" --sizelimit "${ROOT_LENGTH}" "${IMG_FILE}")
`

I added the sleep 5 to give some time for the disk to sync before the losetup.

PhistterbutInc, 2019-08-07:
Where in the script do these lines go?

simplextech, 2019-08-07:
The `sleep 5` is all that I added.  Place it right about the two lines I showed previously.

PhistterbutInc, 2019-08-07:
Still fails with losetup error.

simplextech, 2019-08-07:
On my build host which is a VM and using the docker build the 5 seconds was all I needed.  Depending on your system you may need to pause longer.  Or maybe this is completely irrelevant and was a fluke and without other changes just started working on my system.

PhistterbutInc, 2019-08-07:
I'll try it with sleep 30 and let you know. Doesn't make much sense though, still, that it's reporting no folder to put the file in.

simplextech, 2019-08-07:
I just re-ran on my test box and it's failing now again.  I agree it doesn't make any sense, but for whatever reason it did work and built several images that night.  Now things are failing again.  No changes at all, booted the VM and it's failing to build again.

hhromic, 2019-08-08:
I tested a Docker-based build just now with a fresh `pigen` checkout on a standard Debian Stretch host over baremetal (no VMs). All is working fine here. I build images regularly with Docker and I have never seen this problem before.

```
Device                                                                                    Boot  Start     End Sectors  Size Id Type
/pi-gen/work/2019-08-08-raspbian-buster/export-image/2019-08-08-raspbian-buster-lite.img1        8192  532480  524289  256M  c W95 FAT32 (LBA)
/pi-gen/work/2019-08-08-raspbian-buster/export-image/2019-08-08-raspbian-buster-lite.img2      540672 4300799 3760128  1.8G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

/boot: offset 4194304, length 268435968
/:     offset 276824064, length 1925185536
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mke2fs 1.44.5 (15-Dec-2018)
mount: /dev/loop1 mounted on /pi-gen/work/2019-08-08-raspbian-buster/export-image/rootfs.
mount: /dev/loop0 mounted on /pi-gen/work/2019-08-08-raspbian-buster/export-image/rootfs/boot.
[11:20:28] End /pi-gen/export-image/prerun.sh
```

Perhaps you can give more details about your host system?

If there is indeed a race condition (subject to confirmation), the best approach would be to make a simple wait routine for the `lo` devices to be setup before trying to mount them (instead of `sleep`ing).

simplextech, 2019-08-08:
I think the key difference here is the VM overlay with the docker-build.  I again had the losetup failure even with the simple sleep statement.  So I removed the sleep and re-ran and for whatever reason then it completed without the problem.  So this is a transient issue that I haven't nailed down yet.

PhistterbutInc, 2019-08-08:
@hhromic to reiterate "I'm using a VMware install of Ubuntu 18.04lts"

PhistterbutInc, 2019-08-10:
The issue has also happened on a clean VMware install of Debian 10 Gnome

SvenDowideit, 2019-08-28:
I just had the same happen on my notebook running Debian and DOcker 19.03

artynet, 2019-09-07:
please load the `loop` kernel module before launching the build. In all  the pure Debian distributions is not loaded by default at startup.

Edit your **/etc/modules** file so to have something just like that :

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

binfmt_misc

loop
```
or at runtime run :`

```
$ modprobe loop
$ modprobe binfmt_misc
```

then relaunch the build. That should fix it.

ooxi, 2019-09-22:
This really seems like a timing problem. The script should be made aware of it and wait (with a timout) instead of assuming it just work or the user has loaded some kernel module beforehand

arky, 2019-10-26:
This problem could be reproduced both on Ubuntu 19.04 and 19.10 64bit machines. Any solution with this loopback device timing issue would be much appreciated.

XECDesign, 2019-10-28:
Default 19.10 install. Only added pi-gen, its dependencies and a patch to fix another issue.
Built without any issues as far as I can tell.

der-eismann, 2020-01-16:
We had this problem on our CentOS 7/8 worker machines. We ended up generating loop devices at boot, this is working fine so far. The modules were loaded, but it seemed like the docker container was unable to create the devices. If they existed already there was no problem.
```
#!/bin/bash

set -xeuo pipefail

for i in {0..5}
do
    if [ ! -b /dev/loop${i} ]; then
        mknod -m 0660 /dev/loop${i} b 7 ${i}
    fi
done

chown root:disk /dev/loop*
```