RPi-Distro/pi-gen #325: Build fails when removing things from /boot/cmdline.txt

mprzygrodzki, 2019-08-22:
Hello,

Due to purposes of the devices attached to my RaspberryPi 3B, i need to disable login shell to be accessible over serial. I've found, that after disabling it from raspi-config it's removing from /boot/cmdline.txt following entry: "console=serial0,115200" which is fine and everything works as expected.

But I would like to have management shell over serial disabled during my build with new firmware on SD card.

I've removed entries "console=serial0,115200" from /stage1/00-boot-files/files/cmdline.txt and from /stage2/00-patches/07-resize-init.diff - but after this builds are failing with this kind of message:

`"Applying patch /home/ubuntu/biot-rpi-probiot/stage2/01-sys-tweaks/00-patches/07-resize-init.diff", "patching file rootfs/boot/cmdline.txt", "patch unexpectedly ends in middle of line", "patch: **** malformed patch at line 4:  ",`

Can anybody advice me how to solve this issue ?

XECDesign, 2019-08-22:
I usually manage the patches using quilt. Drop a file named `EDIT` into `/home/ubuntu/biot-rpi-probiot/stage2/01-sys-tweaks/00-patches/`, then when the build gets to it, it will drop you into a shell to fix things up. You can then force push the patch, edit the file by hand and run `quilt refresh` to update it. Then `exit 0` to continue the build. If that all works, you can delete the .rej and EDIT files.

hhromic, 2019-08-22:
@mprzygrodzki the error you are getting is the patch file `stage2/01-sys-tweaks/00-patches/07-resize-init.diff` not being able to be applied anymore because you naturally modified the file being patched: `boot/cmdline.txt`.

For simplicity, I would recommend you to just update the patch for your change.

The patch file originally is this (containing `console=serial0,115200`):
```
--- stage2.orig/rootfs/boot/cmdline.txt
+++ stage2/rootfs/boot/cmdline.txt
@@ -1 +1 @@
-console=serial0,115200 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
+console=serial0,115200 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```

So modify it to not expect `console=serial0,115200` anymore:
```
--- stage2.orig/rootfs/boot/cmdline.txt
+++ stage2/rootfs/boot/cmdline.txt
@@ -1 +1 @@
-console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
+console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```

And it should work. Fingers crossed!

hhromic, 2019-08-22:
@XECDesign thanks btw for your tip.
I never used quilt before but looks very practical indeed if it works like you describe. Cheers!

mprzygrodzki, 2019-08-22:
Will check both solutions, thanks guys.

mprzygrodzki, 2019-08-23:
@hhromic - when I only remove content from the patch, leaving cmdline unchanged i'm receiving following error:

```
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file rootfs/boot/cmdline.txt
```

hhromic, 2019-08-23:
You have to remove the content in **both** places, the `cmdline.txt` file and the patch file. This way the patching program can find the exact line it needs to patch.

mprzygrodzki, 2019-08-23:
@hhromic, so after applying changes to both files i'm getting same error as in my first post:

```
Applying patch /root/biot-rpi-probiot/stage2/01-sys-tweaks/00-patches/07-resize-init.diff
patching file rootfs/boot/cmdline.txt
patch unexpectedly ends in middle of line
patch: **** malformed patch at line 4:

Patch /root/biot-rpi-probiot/stage2/01-sys-tweaks/00-patches/07-resize-init.diff does not apply (enforce with -f)
```
Files are looking like this:

stage1/00-boot-files/files/cmdline.txt
`dwc_otg.lpm_enable=0 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

stage2/01-sys-tweaks/00-patches/07-resize-init.diff
```
--- a/rootfs/boot/cmdline.txt
+++ b/rootfs/boot/cmdline.txt
@@ -1 +1 @@
-dwc_otg.lpm_enable=0 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
+dwc_otg.lpm_enable=0 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```

hhromic, 2019-08-23:
I did a quick test and it worked for me:
```
$ curl -SLO https://github.com/RPi-Distro/pi-gen/raw/master/stage1/00-boot-files/files/cmdline.txt
$ curl -SLO https://github.com/RPi-Distro/pi-gen/raw/master/stage2/01-sys-tweaks/00-patches/07-resize-init.diff
(.. removed `console=serial0,115200` from both files ..)
$ patch -p0 < 07-resize-init.diff
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- stage2.orig/rootfs/boot/cmdline.txt
|+++ stage2/rootfs/boot/cmdline.txt
--------------------------
File to patch: cmdline.txt
patching file cmdline.txt
```

However I went to google about the actual error you are getting:

    "patch unexpectedly ends in middle of line"

And this [answer](https://unix.stackexchange.com/a/1403) in StackExchange suggests that you might have line-ending issues (CRLF being added). Did you edit these files with the same editor and make sure both have the same line endings?

Good luck.

mprzygrodzki, 2019-08-23:
I'm not using the latest branch, need to use stretch for some reasons. 

Will check the suggestion with CRLF, that might be the issue.

matthijskooijman, 2019-10-11:
I've also run into problems with modifying `cmdline.txt`, adding a patch in a custom stage. Problems I've seen is that:
 - When rebasing on top of latest master that also modifies `cmdline.txt`, my patches start to fail and need to be refreshed, which is annoying.
 - When making multiple modifications in `cmdline.txt` (because I have multiple, logically separate changes), I can no longer easily re-order those steps (or make a new custom image where I include the second change, but not the first).

One solution for this that springs to mind, would be to support a `cmdline.txt.d` where multiple files each with one, or a related set of cmdline arguments are stored. I assume the bootloader that reads the file at runtime does not support this, so this directory should be collapsed into a single `cmdline.txt` file at the end, probably in the `export-image` stage (simply by concatenating all files in sorted order, separated by whitespace).

boardofchris, 2020-02-02:
I am experiencing related issues with quilt. All patches are applied the first time pi-gen is run. Runs not clean fail to apply some patches.

The best I can determine, is files from previous stages are copied forward by pi-gen. When quilt runs in a later stage, it finds the previous quilt .pc data indicating the patch was applied, but the impacted files were overwritten by the pi-gen stage progression, reversing the patch. The end result is quilt does not apply the patches and the resulting image is missing patches (broken).

It seems pi-gen should not use quilt in this way.

mprzygrodzki, 2022-04-05:
Late but fixed, the issue on my side was as mentioned, trailing whitespaces.