RPi-Distro/pi-gen #327: Real-time kernel builds?

JonnyTech, 2019-08-30:
Is it possible to create images containing a real-time kernel?

XECDesign, 2019-08-30:
I don't see why not.

JonnyTech, 2019-08-30:
Excellent! Which settings need to be applied? I cannot find anything within the documentation.

XECDesign, 2019-08-30:
There's nothing specific, you'll need to be general familiarity with pi-gen and what you want to do exactly. The existing stages have plenty of example showing how to copy files, execute scripts, add repos, install packages and so on. There's no "install custom kernel" setting or anything like that.

artynet, 2019-09-07:
@JonnyTech check out this repo :

https://github.com/hypriot/rpi-kernel

I use customised version of these scripts to build my own kernel DEBS, they're quite simple to tweak....

JonnyTech, 2019-09-07:
Thanks all

closed 2019-09-07 by JonnyTech