RPi-Distro/pi-gen #332: setting custom root password

artynet, 2019-10-02:
I'd like to change the default root password from `root` to the `FIRST_USER_PASS` value. Anyway, the changes are not registered during the build process...

```patch
diff --git a/stage1/01-sys-tweaks/00-run.sh b/stage1/01-sys-tweaks/00-run.sh
index 9c380ec..c60340f 100755
--- a/stage1/01-sys-tweaks/00-run.sh
+++ b/stage1/01-sys-tweaks/00-run.sh
@@ -9,7 +9,7 @@ if ! id -u ${FIRST_USER_NAME} >/dev/null 2>&1; then
        adduser --disabled-password --gecos "" ${FIRST_USER_NAME}
 fi
 echo "${FIRST_USER_NAME}:${FIRST_USER_PASS}" | chpasswd
-echo "root:root" | chpasswd
+echo "root:${FIRST_USER_PASS}" | chpasswd
 EOF

```

any hints on that ?

XECDesign, 2019-10-02:
It gets unset here:
https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/01-run.sh#L53

artynet, 2019-10-04:
Perfect, it did the trick ! I will close the issue. Thank you very much again !

closed 2019-10-04 by artynet

rpier1, 2020-03-04:
> It gets unset here:
> https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/01-run.sh#L53

Yes but why set the password of root to "root" in the first place?

https://github.com/RPi-Distro/pi-gen/blob/08fc0b9a8237c5a9926a0e9d660156a1e838d33a/stage1/01-sys-tweaks/00-run.sh#L12

only to disable it later

https://github.com/RPi-Distro/pi-gen/blob/08fc0b9a8237c5a9926a0e9d660156a1e838d33a/stage2/01-sys-tweaks/01-run.sh#L53

PS: 
Does `usermod --pass='*' root`   
a) really disable the password, 
or does it 
b) just set the password to something random
?

If a) ... how can one verify that the password is really disabled? Thanks.