RPi-Distro/pi-gen #336: Failure around openssh-server, ssh configuration at Stage2

alekseichuk, 2019-10-22:
Build results in `openssh-server` and `ssh` packages installation failure. 
```
Running hooks in /etc/ca-certificates/update.d...
done.
Processing triggers for dbus (1.12.16-1) ...
Errors were encountered while processing:
 openssh-server
 ssh
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Previous stages have been built with no issues. 
In execution log I see the it actually fetches binaries and installs them but at some point later, during configuration phase, an error 255 is returned which in its orders prevents ssh from being properly installed.

```
Get:45 http://raspbian.mirror.uk.sargasso.net/raspbian buster/main armhf openssh-server armhf 1:7.9p1-10+deb10u1 [291 kB]

...

Selecting previously unselected package openssh-server.
Preparing to unpack .../007-openssh-server_1%3a7.9p1-10+deb10u1_armhf.deb ...
Unpacking openssh-server (1:7.9p1-10+deb10u1) ...
Selecting previously unselected package ssh.
Preparing to unpack .../008-ssh_1%3a7.9p1-10+deb10u1_all.deb ...
Unpacking ssh (1:7.9p1-10+deb10u1) ...

...

Setting up openssh-server (1:7.9p1-10+deb10u1) ...

Creating config file /etc/ssh/sshd_config with new version
qemu: Unsupported syscall: 382
qemu: Unsupported syscall: 382
Creating SSH2 RSA key; this may take some time ...
qemu: Unsupported syscall: 384
PRNG is not seeded

dpkg: error processing package openssh-server (--configure): installed openssh-server package post-installation script subprocess returned error exit status 255

...

dpkg: dependency problems prevent configuration of ssh:
 ssh depends on openssh-server (>= 1:7.9p1-10+deb10u1); however:
  Package openssh-server is not configured yet.
```

As a way to solve the issue, I went through the standart steps e.g. mentioned [here](https://itsfoss.com/dpkg-returned-an-error-code-1) but none of them helped.

Regarding the machine, it's VirtualBox with Linux Ubuntu 16.04 LTS 32-bit. Last time I used it, roughly a month ago, it built the image without any problems.

Full build log captured from terminal: 
https://gist.github.com/alekseichuk/914d19a25133fee1e1e1455e43d1645f

XECDesign, 2019-10-22:
Have you tried using a newer version of qemu-user-static?

alekseichuk, 2019-10-22:
@XECDesign I haven't, just use all what comes by default. Will try updating it now.

alekseichuk, 2019-10-22:
Invoked: `sudo apt-get install qemu binfmt-support qemu-user-static`. 
Unfortunately, ended up with the same error.

XECDesign, 2019-10-22:
I meant a version newer than what's available in 16.04. Like qemu 4.0 - https://packages.ubuntu.com/eoan/qemu-user-static

alekseichuk, 2019-10-23:
Hi @XECDesign So, I tried installing qemu 4.0 from the given link yesterday. I'm not sure wheter it was installed properly, as for the installation I used just `.deb` file and GUI installer for packages and for some reason after oppening the file again the tool continued showing "Install" button. Given all that uncertainty, I tried building again and it has failed with exactly the same error. 

So, eventually I downloaded Xubuntu 18.04 (as there's no 32-bit version of Ubuntu 18) and everything returned back to normal. Managed to bake my build. Though it's a bit weird why all of a sudden my previous setup broke down.

Anyway, presume the issue can be closed. 
Thank you very much for running the project and helping us.

closed 2019-10-24 by alekseichuk

XECDesign, 2019-10-24:
Thanks, Alek. We might be getting to a point where we should just bundle a qemu-arm-static binary with pi-gen, since the version is starting to matter.