RPi-Distro/pi-gen #340: install: target '/media/pi/SSD/pi-gen/work/2019-11-06-Raspbian/stage0/rootfs/etc/apt/' is not a directory: No such file or directory

DominoX, 2019-11-06:
Hi @XECDesign 

Please refer #338 for my setup

Following your advice there, `sudo ./build.sh |& sudo tee build.log` now yields a different error almost immediately. build.log contains:

`[11:36:08] Begin /media/pi/RP4-Data/pi-gen`
`[11:36:08] Begin /media/pi/RP4-Data/pi-gen/stage0`
`[11:36:08] Begin /media/pi/RP4-Data/pi-gen/stage0/prerun.sh`
`[11:36:08] End /media/pi/RP4-Data/pi-gen/stage0/prerun.sh`
`[11:36:08] Begin /media/pi/RP4-Data/pi-gen/stage0/00-configure-apt`
`[11:36:08] Begin /media/pi/RP4-Data/pi-gen/stage0/00-configure-apt/00-run.sh`
`install: target '/media/pi/RP4-Data/pi-gen/work/2019-11-06-Raspbian/stage0/rootfs/etc/apt/' is not a directory: No such file or directory`

Thanks again for all you do :+1:

XECDesign, 2019-11-06:
> The /media/pi/RP4-Data device I'm using is a 128GB SSD plugged into USB. Filesys = ext4. Auto-mounted by Buster when inserted, so permissions are default.

I'm not sure that the permissions are ok. Automounted stuff tend to be cautious and adds options like 'nodev' will break pi-gen.

Try unmounting and mounting it manually. Check the output of `mount` to make sure it's not using something like `noexec` or `nodev`

> `sudo ./build.sh |& sudo tee build.log`

This isn't particularly relevant, but you don't need that second `sudo`. Understanding the commands you're typing goes a long way towards making sure everything goes smoothly.

XECDesign, 2019-11-06:
Also try `sudo CLEAN=1 ./build.sh |& tee build.log`. Remove the old `build.log` first, since it will be owned by root (because you used sudo when it wasn't necessary).

DominoX, 2019-11-06:
> > `sudo ./build.sh |& sudo tee build.log`
> 
> This isn't particularly relevant, but you don't need that second `sudo`. Understanding the commands you're typing goes a long way towards making sure everything goes smoothly.

Yeah, I wondered about that. But without that 2nd `sudo`, i get `tee: build.log: Permission denied` in the first line of the output, just above all that other stuff in my IP. 

Seems like your hunch about auto-mount permissions might be correct. The tee error above again points to some sort of permissions problem.

XECDesign, 2019-11-06:
Probably because everything on the card is owned by root, so normal users don't have write permissions. Heading toward general linux usage territory.

DominoX, 2019-11-06:
Yes, everything owned by root because that's what the auto-mount does. Am I missing some documentation or tutorial somewhere where these pre-requisites are listed (like not relying on auto-mount but mounting manually). I'd hate to be wasting your time if I just missed a simple setup guide somewhere?

Best, Dave

XECDesign, 2019-11-06:
> Yes, everything owned by root because that's what the auto-mount does.

Automount wouldn't be changing ownership.

>  I just missed a simple setup guide somewhere?

No, I think it just required a bit more familiarity with linux. You'd be running into the same sort of problems with any other software that did similar things.

I wouldn't mind going through everything step by step, but I'm a little busy with work right now and github issues are a bad medium for this.

revmischa, 2020-01-26:
Are you running this in an encrypted home folder? I was and got this error. Try making /pi-build or something and run it there

Kudesnick, 2020-10-03:
Hello!

I have this error too:
```
sudo ./build.sh
[01:19:53] Begin /mnt/d/git/st/pi-gen
[01:19:53] Begin /mnt/d/git/st/pi-gen/stage0
[01:19:53] Begin /mnt/d/git/st/pi-gen/stage0/prerun.sh
[01:19:53] End /mnt/d/git/st/pi-gen/stage0/prerun.sh
[01:19:53] Begin /mnt/d/git/st/pi-gen/stage0/00-configure-apt
[01:19:53] Begin /mnt/d/git/st/pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/mnt/d/git/st/pi-gen/work/2020-10-04-CrazyPi/stage0/rootfs/etc/apt/' is not a directory: No such file or directory
```
But I'm using ws2 for windows 10 version 2004 (OS Build 19041.508)
I create manually folders "etc" and "get", but I get error:
```
sudo ./build.sh
[01:33:21] Begin /mnt/d/git/st/pi-gen
[01:33:21] Begin /mnt/d/git/st/pi-gen/stage0
[01:33:21] Begin /mnt/d/git/st/pi-gen/stage0/prerun.sh
[01:33:21] End /mnt/d/git/st/pi-gen/stage0/prerun.sh
[01:33:21] Begin /mnt/d/git/st/pi-gen/stage0/00-configure-apt
[01:33:21] Begin /mnt/d/git/st/pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/mnt/d/git/st/pi-gen/work/2020-10-04-CrazyPi/stage0/rootfs/etc/apt/sources.list.d/' is not a directory: No such file or directory
```
Host filesystem is NTFS.

DominoX, 2022-03-07:
No longer relevant

closed 2022-03-07 by DominoX

galkinvv, 2023-07-13:
For anybody reading this issue with having error like
```install: cannot create regular file '/path/to/pi-gen/work/name/stage0/rootfs/etc/apt/': No such file or directory``` 

This comment helped me https://github.com/RPi-Distro/pi-gen/issues/340#issuecomment-550485338
It seems that problem was caused by some leftovers from a previous failed try, so using `sudo CLEAN=1` instead of just `sudo` resolved this issue

piotrbetlej, 2025-06-24:
My problem were mount options of the working partition. Setting to default rw,relatime did the job.