RPi-Distro/pi-gen #346: Document or support environment clean-up

erichiggins, 2019-11-16:
Hello!  I've just started using pi-gen and have had varying levels of success. 

One thing I think would be enormously helpful is a better understanding of **how to clean up/start over**. The docs aren't clear about this, there is no `./clean.sh` command (for example), and my best guess is based on the contents of your `.gitignore` file -- to simply delete the contents of `deploy/` and `work/`.

Is there more to it than that? Could the docs be improved, or better, a script added to do this work? Thanks!

XECDesign, 2019-11-16:
You'd just make sure nothing from the pi-gen directory is mounted and then remove the directories you mentioned. You can also use CLEAN=1 to do a clean build, but it won't remove files from previous dates.

erichiggins, 2019-11-16:
Ok, but if I do multiple builds on the same date, then they would overlap and not be cleaned -- is that right?

XECDesign, 2019-11-16:
If you use CLEAN=1 it will delete the files from each stage before starting and then overwrite the files in deploy/

erichiggins, 2019-11-16:
Got it. So then I guess I have two requests, because I think it's important to separate those steps and make it easy to understand and verify the behavior.

1. Clarify this behavior in the README
2. Add an optional, stand-alone clean up command