RPi-Distro/pi-gen #350: Silent failures

erichiggins, 2019-11-18:
My build is based off of the `2019-07-10-raspbian-buster` branch/tag.

There are a few commands I'm running in my `01-run.sh` within `stage2` which seem to cause the build to fail silently.

Here's the specific block, for reference, but I'm happy to create a branch/PR/whatever to surface all of my changes if needed.

```
on_chroot <<EOF
cd /home/pi/projects/trickler/peripheral/
sudo -u pi pm2 start ecosystem.config.js
sudo -u pi pm2 startup systemd
sudo env PATH=$PATH:/usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/bin /usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
sudo -u pi pm2 save
EOF

echo "DONE WITH RUN"
```

The last few lines of the typical output I get is:

```
sudo: unable to resolve host 69a9ef18b5c5: Temporary failure in name resolution
[PM2] Init System found: systemd
[PM2] To setup the Startup Script, copy/paste the following command:
sudo env PATH=$PATH:/usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/bin /usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi

real	7m53.294s
user	0m0.273s
sys	0m0.385s
```

But if I remove all of the chroot command lines after `sudo -u pi pm2 start ecosystem.config.js`, then I get this (which also seems like a failure):

```
DONE WITH RUN
[01:49:04] End /pi-gen/stage2/04-opentrickler/01-run.sh
[01:49:04] End /pi-gen/stage2/04-opentrickler
umount: /pi-gen/work/2019-11-17-opentrickler/stage2/rootfs/proc: target is busy.

real    7m44.145s
user    0m0.291s
sys     0m0.376s
```

My setup is not complicated. I'm installing a specific version of NodeJS, some dependencies, and my application code. I'm running the Docker build inside a VirtualBox VM. I tried just the VM first but the image was getting really contaminated (`/home/vagrant` was being copied over?!) and the environment ended up breaking entirely.

Any help would be greatly appreciated. I've spent about 7 days messing with pi-gen and have yet to get a workable image.

erichiggins, 2019-11-18:
In case it helps, here's the full diff of my changes.
https://gist.github.com/erichiggins/85051a51412ed0cc9a6f30feacce8354

RaspberryPiFan, 2019-11-26:
`umount: /pi-gen/work/2019-11-17-opentrickler/stage2/rootfs/proc: target is busy.`
That's no big problem, that happens sometimes

RaspberryPiFan, 2019-11-26:
@erichiggins Please give me your complete pi-gen folder so I can help better.

erichiggins, 2019-11-27:
@RaspberryPiFan The gist in my comment above contains the full diff that should allow you to reproduce, I believe. If there are other details you need, let me know which ones specifically and I"ll see if I can grab them.

Re: the umount error: I got this several times and it never seemed to resolve itself. Perhaps some retry logic could be added to the build so that it won't fail on that? Not sure what the root cause is.

yeppao, 2020-05-09:
@erichiggins all you need to do is kill the god daemon of PM2.

```
on_chroot <<EOF
cd /home/pi/projects/trickler/peripheral/
sudo -u pi pm2 start ecosystem.config.js
sudo -u pi pm2 startup systemd
sudo env PATH=$PATH:/usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/bin /usr/local/lib/nodejs/node-v8.16.2-linux-armv6l/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
sudo -u pi pm2 save
sudo -u pi pm2 kill
EOF

echo "DONE WITH RUN"
```

And voila