RPi-Distro/pi-gen #358: Docker build LOOP_SET_STATUS64 failed: Resource temporarily unavailable

jtc42, 2019-12-05:
Hi,

I'm having trouble getting docker builds to work. I've modified export-image/presun.sh to give extra debug information around the issue, resulting in:

```
7509 Syncing disks.
7510 11780: loopdev:      CXT: [0xfffc84c4]: initialize context
7511 11780: loopdev:      CXT: [0xfffc84c4]: init: ignore ioctls
7512 11780: loopdev:      CXT: [0xfffc84c4]: init: loop-control detected 
7513 11780: loopdev:      CXT: [0xfffc84c4]: find_unused requested
7514 11780: loopdev:      CXT: [0xfffc84c4]: using loop-control
7515 11780: loopdev:      CXT: [0xfffc84c4]: loop2 name assigned
7516 11780: loopdev:      CXT: [0xfffc84c4]: find_unused by loop-control [rc=0]
7517 11780: loopdev:      CXT: [0xfffc84c4]: set offset=4194304
7518 11780: loopdev:      CXT: [0xfffc84c4]: set sizelimit=268435968
7519 11780: loopdev:      CXT: [0xfffc84c4]: set backing file=/pi-gen/work/2019-12-05-raspbian-openflexure-buster/export-imag
7520 11780: loopdev:    SETUP: [0xfffc84c4]: device setup requested
7521 11780: loopdev:    SETUP: [0xfffc84c4]: backing file open: OK
7522 11780: loopdev:      CXT: [0xfffc84c4]: open /dev/loop2 [rw]: Success
7523 11780: loopdev:    SETUP: [0xfffc84c4]: device open: OK
7524 11780: loopdev:    SETUP: [0xfffc84c4]: LOOP_SET_FD: OK
7525 11780: loopdev:    SETUP: [0xfffc84c4]: LOOP_SET_STATUS64 failed: Resource temporarily unavailable
7526 11780: loopdev:    SETUP: [0xfffc84c4]: failed [rc=-11]
7527 losetup: /pi-gen/work/2019-12-05-raspbian-buster/export-image/2019-12-05-raspbian-buster-full.img: failed to set up loop device: Resource temporarily unavailable
7528 11780: loopdev:      CXT: [0xfffc84c4]: de-initialize
7529 11780: loopdev:      CXT: [0xfffc84c4]: closing old open fd
7530 11780: loopdev:     ITER: [0xfffc8644]: de-initialize
```

Important bit seems to be `LOOP_SET_STATUS64 failed: Resource temporarily unavailable`.

This seems vaguely related to https://github.com/RPi-Distro/pi-gen/issues/320, but I've found nothing else saying `Resource temporarily unavailable`.

This occurs when running  the docker build in a VM.

I've checked and all the required kernel modules are enabled.

Any thoughts or advice would be appreciated.

Thanks

jtc42, 2019-12-06:
I think it's related to this change:
https://github.com/torvalds/linux/commit/5db470e229e22b7eda6e23b5566e532c96fb5bc3

I have a feeling that for some reason `losetup` is trying to mount the disk in the middle of a sync operation, and giving this error. It's returning error `EAGAIN`, which as far as I know means that the operation should work eventually, but currently pi-gen has no mechanism to wait until this time, and so just exits with an error.

KeithPetro, 2020-01-16:
Has anyone found a workaround to this issue? I'm currently testing using `kpartx` to see if that will work around this issue.

jtc42, 2020-01-17:
> Has anyone found a workaround to this issue? I'm currently testing using `kpartx` to see if that will work around this issue.

We seem to be getting stable builds by patching `export-image/prerun.sh` like so:
https://gitlab.com/openflexure/pi-gen/blob/8f2539d4f701c623d888a435da418991a411ef5f/export-image/prerun.sh

Lines 44 and 50 are the important ones. For us at least, it appeared to be a timing issue, so we basically just let it continually try the offending `losetup` command until it works. So far I haven't seen it need to retry more than once so fingers crossed.

Course, it could break again. This is maybe the 4th time I've thought we fixed it, only to have the problem come back. This version seems pretty stable though.

KeithPetro, 2020-01-18:
> We seem to be getting stable builds by patching `export-image/prerun.sh` like so:
> https://gitlab.com/openflexure/pi-gen/blob/8f2539d4f701c623d888a435da418991a411ef5f/export-image/prerun.sh
>
> Lines 44 [...]

We also _think_ we've fixed the issue by running headless. Before we were using a full gnome build of Ubuntu, and it seems like the gnome loop devices were causing it grief. `losetup` _should_ be able to make the new devices as needed, but it doesn't seem to...

jtc42, 2020-01-18:
Okay great, let me know how it goes. As I mentioned we had a couple of false hope situations like that before we discovered the kernel change causing our “resource temporarily unavailable” error, and sticking our botch-fix loop in.

The weirdest thing is both of our build servers are running different versions of Ubuntu Server, in very different environments, and yet experienced the exact same issue. I never tried with basic Debian, but I wonder if that would have helped.

cshoredaniel, 2020-01-18:
I've seen this on basic Debian with Xfce desktop (so gvfs-bin and a few gnomisms of that sort).

KeithPetro, 2020-01-21:
> Okay great, let me know how it goes. As I mentioned we had a couple of false hope situations like that before we discovered the kernel change causing our “resource temporarily unavailable” error, [...]

Since that post we've only had successful builds. Fingers crossed that that's all that was needed. Just make sure that there are no pesky gnome snap services mounted on the loop devices (if there are, unmount them in headless mode).

gklp, 2020-06-11:
Hello @KeithPetro @jtc42 . Did you solve this issue in locally ?

jtc42, 2020-06-11:
> Hello @KeithPetro @jtc42 . Did you solve this issue in locally ?

We solved it in a kind of nasty way, but it's worked reliably ever since:
https://gitlab.com/openflexure/pi-gen/blob/8f2539d4f701c623d888a435da418991a411ef5f/export-image/prerun.sh#L44

gklp, 2020-06-11:
> > Hello @KeithPetro @jtc42 . Did you solve this issue in locally ?
> 
> We solved it in a kind of nasty way, but it's worked reliably ever since:
> https://gitlab.com/openflexure/pi-gen/blob/8f2539d4f701c623d888a435da418991a411ef5f/export-image/prerun.sh#L44

Thank you @jtc42 . I've tested it. Sometimes, it is working sometimes not. But finally I created an image ! Thanks again.

XECDesign, 2020-09-24:
Fixed by https://github.com/RPi-Distro/pi-gen/pull/449/

closed 2020-09-24 by XECDesign