RPi-Distro/pi-gen #361: /stage0/rootfs/proc: proc already mounted on /proc.

jangrewe, 2019-12-06:
Still the same setup as !360, and now i do see that the base system gets installed, but then it suddenly stops:

```
[...]
I: Configuring libc-bin...
I: Configuring systemd...
I: Base system installed successfully.
[18:41:12] End /pi-gen/stage0/prerun.sh
[18:41:12] Begin /pi-gen/stage0/00-configure-apt
[18:41:12] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
/stage0/rootfs/proc: proc already mounted on /proc.

real    8m7.208s
user    0m0.123s
sys     0m0.091s
```

XECDesign, 2019-12-09:
The safest thing to do would be to reboot and try again with CLEAN=1. Does that make any difference or do you always get stuck there?

matthijskooijman, 2019-12-11:
Could it be that a previous aborted run still left proc mounted?