RPi-Distro/pi-gen #363: Too simple WPA_PASSWORD aborts script

julled, 2019-12-20:
Hi,

if i try to use WPA_PASSWORD = "AAA"  or "" as password, the script aborts without any further hints that this might be a problem:

> 
> 'files/wpa_supplicant.conf' -> '/pi-gen/work/2019-12-20-piImage/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
> 
> real	0m34,597s
> user	0m0,029s
> sys	0m0,037s

I searched for a while to find this issue.
Maybe a hint in the readme could help against this.

samtygier, 2019-12-23:
So there are a couple of issues here.
1) wpa_passphrase writes its error messages to stdout, so they get written to wpa_supplicant.conf instead of being visible. We could use tee so that it is visible.
2) No early check. It would be simple to check this when reading variables rather than wait for it to reach stage2

I'll prep some patches for each.

samtygier, 2019-12-26:
> wpa_passphrase writes its error messages to stdout

This is now fixed upstream, but it would be good to fix it here as it will take a while for the change to reach the raspbian repos. https://w1.fi/cgit/hostap/commit/?id=32289112c40ba3f6d0b3c04263654d6223055e22

XECDesign, 2020-01-07:
Thank you both.

closed 2020-01-07 by XECDesign