RPi-Distro/pi-gen #381: Some issues with new HOSTNAME feature

bit-bash, 2020-01-20:
Since the new HOSTNAME feature was added in commits:
   d30343f626916a5b6f23cec67df919bd5a5b643a
   95ba042c8393682eecab36e6c7056a28c5cafed4
I've noticed a few minor-ish issues.

First, in stage1/02-net-tweaks/00-run.sh, the line:
echo "127.0.1.1		${HOSTNAME}" >> "${ROOTFS_DIR}/etc/hosts"
has two tabs - it should be only ONE tab to be consistent with previous builds and to prevent regressions with derivative repos (such as teslausb which expects one tab for its patch process).

Second, if build.sh is run multiple times, /etc/hosts will end up with multiple 127.0.1.1... lines.  Each time build.sh is run a new line like this will be added to /etc/hosts.  It might be prudent for the shell script to check to see if a 127.0.1.1 line already exists before appending a new one.

Thirdly, when building with the recommended ubuntu 16.04 32-bit distro, HOSTNAME already exists by the OS.  Therefore, the default of "raspberrypi" is not quite true in the README.md file.  I had to add HOSTNAME=raspberrypi to my config file otherwise the hostname of the built target took on the name of my ubuntu 16.04 host.

Oh, and slightly off-topic, to get everything to build with ubuntu 16.04 32-bit, I had to install:
   qemu-user-static_4.0+dfsg-0ubuntu9.2_i386.deb
   binfmt-support_2.2.0-2_i386.deb
from ubuntu eoan-updates.  The default qemu-user-static package for ubuntu 16.04 is too old to handle debian buster images.

XECDesign, 2020-01-20:
Thanks for the helpful bug report.

bit-bash, 2020-01-22:
Looks like issue #3 has been addressed by https://github.com/RPi-Distro/pi-gen/commit/24bb01dec280f568ac59cdd1253a33a6d30ac4c4

Yay!