RPi-Distro/pi-gen #385: New 2020-02-05 image has Wireless LAN Soft blocked even if wpa supplicant has country setting

guysoft, 2020-02-16:
Hey,
Spotted a change between ``2019-09-26`` and ``2020-02-05``, the rfkill shows with my wpa-supplicant setting:

```
 sudo rfkill list
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

However this setup works fine on ``2019-09-26``.

**Has anything been changed?**
Because I see nothing in the release notes or looking in the systemd service.

I have yet to try ``2020-02-14`` and will later today since it just came out yesterday. But was wondering if you have any more insight since its evidently something undocumented AFIK.

The wifi setting is identical (the one in OctoPi).
Will note that we create  a wpasuppliant.conf file in ``/boot`` and symlink to the right location in ``/etc/wpa_supplicant/wpa_supplicant.conf``. And that this has worked in all previous versions.

```
# Use this file to configure your wifi connection(s).
#
# Just uncomment the lines prefixed with a single # of the configuration
# that matches your wifi setup and fill in SSID and passphrase.
#
# You can configure multiple wifi connections by adding more 'network'
# blocks.
#
# See https://linux.die.net/man/5/wpa_supplicant.conf
# (or 'man -s 5 wpa_supplicant.conf') for advanced options going beyond
# the examples provided below (e.g. various WPA Enterprise setups).
#
# !!!!! HEADS-UP WINDOWS USERS !!!!!
#
# Do not use Wordpad for editing this file, it will mangle it and your
# configuration won't work. Use a proper text editor instead.
# Recommended: Notepad++, VSCode, Atom, SublimeText.
#
# !!!!! HEADS-UP MACOSX USERS !!!!!
#
# If you use Textedit to edit this file make sure to use "plain text format"
# and "disable smart quotes" in "Textedit > Preferences", otherwise Textedit
# will use none-compatible characters and your network configuration won't
# work!

## WPA/WPA2 secured
network={
  ssid="MY_SSID"
  psk="SECRET_PASSWORD"
}

## Open/unsecured
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#}

## WEP "secured"
##
## WEP can be cracked within minutes. If your network is still relying on this
## encryption scheme you should seriously consider to update your network ASAP.
#network={
#  ssid="put SSID here"
#  key_mgmt=NONE
#  wep_key0="put password here"
#  wep_tx_keyidx=0
#}

# Uncomment the country your Pi is in to activate Wifi in RaspberryPi 3 B+ and above
# For full list see: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
country=GB # United Kingdom
#country=CA # Canada
#country=DE # Germany
#country=FR # France
#country=US # United States

### You should not have to change the lines below #####################

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```

daedeloth, 2020-02-16:
I seem to have the same issue, so following.

XECDesign, 2020-02-17:
That's intentional. Trying to determine whether a country is set is unreliable, so we default to always blocking it by default and then letting the user unblock it. The tools we provide - raspi-config and the GUI applet do this automatically.

guysoft, 2020-02-17:
@XECDesign Is there a headless configuration method you would recommend?
Because all 7 distros I maintain have no access to a terminal or GUI screen.

Also, if you are allowed to share, it might help to understand what requires you to do this, because other distros done have this as a default setting.

XECDesign, 2020-02-17:
In Raspbian, dropping a wpa_supplicant.conf file into the boot directory copies it into /etc/wpa_supplicant/ and removes the block as well

I'm not aware that we are strictly speaking legally required to do it this way, but since we sell hardware, it all gets a bit tricky. I don't think anybody wants to have a precedent-setting case here, so we'll err on the side of caution.

It's in the release notes for the 2019-06-20 image, but the device paths have changed since then and had to be adjusted. Don't expect release notes to be complete. There are thousands of changes between releases and it's not possible to document all of them or know which will be most relevant to mention. Sometimes it seems like the important things are completely ignored, or sometimes things which seem unimportant turn out to be important. It's all easier with the benefit of hindsight after a release.

guysoft, 2020-02-17:
> In Raspbian, dropping a wpa_supplicant.conf file into the boot directory copies it into /etc/wpa_supplicant/ and removes the block as well

Ok, since we have a default one should be sufficient.

> Don't expect release notes to be complete.

I don't I also tried scrolling the commit messages and could not find this. If you can point me to the commit I could reference to it too. Your call on mentioning it. I am just recommending. The fact I am getting multiple issue reports on that from nightly builds is usually a good indicator.

XECDesign, 2020-02-17:
The relevant commit:
https://github.com/RPi-Distro/pi-gen/commit/b4ea63c0454e12af659623d0bd34bba71e6c9f25

guysoft, 2020-02-18:
I guess we can close it.

closed 2020-02-18 by guysoft