RPi-Distro/pi-gen #389: Error: The location 272629760 is outside of the device ../2020-02-24-MyRaspbian/export-image/2020-02-24-MyRaspbian.img

filipecarneiro, 2020-02-25:
I’m trying to add a new custom stage (**stage6**) to the build process.

This stage has several subdirectories but I’ve managed to reproduce the error with just one “00-packages”. 

## My custom stage looks like this:

- Stage6
   - 00-install-packages
      - 00-packages
   - EXPORT_IMAGE
   - EXPORT_NOOBS
   - Prerun.sh

## 00-packages content:

 xdotool unclutter sed
 git
 rfkill
 tree

## Notes

**EXPORT_IMAGE**, **EXPORT_NOOBS** and **prerun.sh** have the same content as the other stages.

I’ve also added **SKIP** empty files to stage3, stage4 and stage5 and **SKIP_IMAGES** to stage4 and stage5 as instructed.

Just to be sure, I’ve specified “**IMAGE_LIST="stage0 stage1 stage2 stage6**" to config.

Is this ok?

### I’ve got this error at the end of the build script:

[18:46:14] Begin /home/fcarneiro/myagent/_work/1/a/export-image/prerun.sh
2020-02-24T18:46:18.6176652Z du: cannot access '/home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/stage6/rootfs': No such file or directory
2020-02-24T18:46:18.6211782Z (standard_in) 1: syntax error
2020-02-24T18:46:18.6216977Z (standard_in) 1: syntax error
2020-02-24T18:46:18.7409080Z Error: The location 272629760 is outside of the device /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/export-image/2020-02-24-MyRaspbian.img.
2020-02-24T18:46:18.7749783Z ##[error]Bash exited with code '1'.
2020-02-24T18:46:18.7776838Z ##[section]Finishing: Build

### The log arround stage6 step has some warnings:

2020-02-24T18:40:55.5058577Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage2/03-set-timezone/02-run.sh
2020-02-24T18:40:55.5150578Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage2/03-set-timezone
2020-02-24T18:40:55.6227243Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage2
2020-02-24T18:40:55.6275791Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage3
2020-02-24T18:40:55.6410751Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage3
2020-02-24T18:40:55.6466787Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage4
2020-02-24T18:40:55.6606170Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage4
2020-02-24T18:40:55.6654765Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage5
2020-02-24T18:40:55.6785636Z [18:40:55] End /home/fcarneiro/myagent/_work/1/a/stage5
2020-02-24T18:40:55.6842856Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage6
2020-02-24T18:40:55.6957906Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage6/00-install-packages
2020-02-24T18:40:55.6990272Z [18:40:55] Begin /home/fcarneiro/myagent/_work/1/a/stage6/00-install-packages/00-packages
2020-02-24T18:40:55.7072256Z **realpath: /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/stage6/rootfs/proc: No such file or directory**
2020-02-24T18:40:55.7133409Z **realpath: /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/stage6/rootfs/dev: No such file or directory**
2020-02-24T18:40:55.7190877Z **realpath: /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/stage6/rootfs/dev/pts: No such file or directory**
2020-02-24T18:40:55.7244725Z **realpath: /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/stage6/rootfs/sys: No such file or directory**
2020-02-24T18:40:55.8819311Z Reading package lists...
2020-02-24T18:40:56.1861540Z Building dependency tree...
2020-02-24T18:40:56.1867889Z Reading state information...
2020-02-24T18:40:56.4918224Z sed is already the newest version (4.4-2).
2020-02-24T18:40:56.4919123Z unclutter is already the newest version (8-21).
2020-02-24T18:40:56.4919779Z xdotool is already the newest version (1:3.20160805.1-3).
2020-02-24T18:40:56.4920391Z git is already the newest version (1:2.17.1-1ubuntu0.5).
2020-02-24T18:40:56.4920495Z The following NEW packages will be installed:
2020-02-24T18:40:56.4920613Z   rfkill tree
2020-02-24T18:40:56.8014934Z 0 upgraded, 2 newly installed, 0 to remove and 3 not upgraded.
2020-02-24T18:40:56.8015902Z Need to get 65.5 kB of archives.
2020-02-24T18:40:56.8016610Z After this operation, 248 kB of additional disk space will be used.
2020-02-24T18:40:56.8018088Z Get:1 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 rfkill amd64 2.31.1-0.4ubuntu3.5 [24.8 kB]
2020-02-24T18:40:56.9122993Z Get:2 http://archive.ubuntu.com/ubuntu bionic/universe amd64 tree amd64 1.7.0-5 [40.7 kB]
2020-02-24T18:40:57.6748122Z debconf: unable to initialize frontend: Dialog
2020-02-24T18:40:57.6749450Z debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
2020-02-24T18:40:57.6750126Z debconf: falling back to frontend: Readline
2020-02-24T18:40:57.6968895Z debconf: unable to initialize frontend: Readline
2020-02-24T18:40:57.6969805Z debconf: (This frontend requires a controlling tty.)
2020-02-24T18:40:57.6970001Z debconf: falling back to frontend: Teletype
2020-02-24T18:40:57.7068128Z dpkg-preconfigure: unable to re-open stdin: 
2020-02-24T18:40:57.7775907Z Fetched 65.5 kB in 0s (144 kB/s)
2020-02-24T18:40:57.9028820Z Selecting previously unselected package rfkill.
2020-02-24T18:40:57.9488822Z (Reading database ... 
2020-02-24T18:40:57.9489389Z (Reading database ... 5%
2020-02-24T18:40:57.9489484Z (Reading database ... 10%
2020-02-24T18:40:57.9489615Z (Reading database ... 15%
2020-02-24T18:40:57.9489779Z (Reading database ... 20%
2020-02-24T18:40:57.9489864Z (Reading database ... 25%
2020-02-24T18:40:57.9489948Z (Reading database ... 30%
2020-02-24T18:40:57.9490031Z (Reading database ... 35%
2020-02-24T18:40:57.9490139Z (Reading database ... 40%
2020-02-24T18:40:57.9490249Z (Reading database ... 45%
2020-02-24T18:40:57.9490330Z (Reading database ... 50%
2020-02-24T18:40:58.0433234Z (Reading database ... 55%
2020-02-24T18:40:58.1403414Z (Reading database ... 60%
2020-02-24T18:40:58.2342681Z (Reading database ... 65%
2020-02-24T18:40:58.3447978Z (Reading database ... 70%
2020-02-24T18:40:58.4425687Z (Reading database ... 75%
2020-02-24T18:40:58.5238106Z (Reading database ... 80%
2020-02-24T18:40:58.6204663Z (Reading database ... 85%
2020-02-24T18:40:58.7676411Z (Reading database ... 90%
2020-02-24T18:40:58.8808330Z (Reading database ... 95%
2020-02-24T18:40:58.8808555Z (Reading database ... 100%
2020-02-24T18:40:58.8808723Z (Reading database ... 86250 files and directories currently installed.)
2020-02-24T18:40:58.8881085Z Preparing to unpack .../rfkill_2.31.1-0.4ubuntu3.5_amd64.deb ...
2020-02-24T18:40:58.9325024Z Unpacking rfkill (2.31.1-0.4ubuntu3.5) ...
2020-02-24T18:40:59.0826118Z Selecting previously unselected package tree.
2020-02-24T18:40:59.0950821Z Preparing to unpack .../tree_1.7.0-5_amd64.deb ...
2020-02-24T18:40:59.1084755Z Unpacking tree (1.7.0-5) ...
2020-02-24T18:40:59.2746751Z Setting up tree (1.7.0-5) ...
2020-02-24T18:40:59.3257541Z Setting up rfkill (2.31.1-0.4ubuntu3.5) ...
2020-02-24T18:40:59.3933146Z Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
2020-02-24T18:41:02.6225732Z [18:41:02] End /home/fcarneiro/myagent/_work/1/a/stage6/00-install-packages/00-packages
2020-02-24T18:41:02.6341695Z [18:41:02] End /home/fcarneiro/myagent/_work/1/a/stage6/00-install-packages
2020-02-24T18:41:02.6407713Z [18:41:02] End /home/fcarneiro/myagent/_work/1/a/stage6
2020-02-24T18:41:02.6459625Z [18:41:02] Begin /home/fcarneiro/myagent/_work/1/a/export-image
2020-02-24T18:41:02.6551027Z [18:41:02] Begin /home/fcarneiro/myagent/_work/1/a/export-image/prerun.sh
2020-02-24T18:41:05.7611277Z /boot: offset 4194304, length 268435456
2020-02-24T18:41:05.7612843Z /:     offset 272629760, length 1572864000
2020-02-24T18:41:05.7698623Z mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
2020-02-24T18:41:05.8474008Z mke2fs 1.44.1 (24-Mar-2018)
2020-02-24T18:41:06.8560358Z mount: /dev/loop3 mounted on /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/export-image/rootfs.
2020-02-24T18:41:06.8657968Z mount: /dev/loop2 mounted on /home/fcarneiro/myagent/_work/1/a/work/2020-02-24-MyRaspbian/export-image/rootfs/boot.
2020-02-24T18:41:23.1071275Z [18:41:23] End /home/fcarneiro/myagent/_work/1/a/export-image/prerun.sh

**Can you please help?**

excitedleigh, 2020-04-21:
Are your scripts (particularly `prerun.sh`) executable? I ran into some similar issues and that turned out to be one of the culprits.

seclorum, 2020-10-01:
This seems to be related to the use of "GB" versus "GiB" in the partition build scripts.

pyr0ball, 2020-10-28:
Any fix on this? I'm currently getting the same issue on the latest master, using the docker image build and a few localization customizations

kmendez, 2023-05-31:
FWIW, I got this exact same error because I didn't pluralize SKIP_IMAGES. I just had empty files named SKIP_IMAGE in the stages I didn't want to run and received the same error. After correcting the name the build process completed without error. 

Hopefully this helps someone along the way.

leodonathilic, 2023-06-14:
edit:
prerun.sh was missing in the stage. 
```
#!/bin/bash -e

if [ ! -d "${ROOTFS_DIR}" ]; then
	copy_previous
fi
```(this should be the content of the prerun.sh file)

It seems to have fixed my problem, maybe it would work too on your side.

dmelo, 2023-10-02:
Had this exact same problem and @kmendez advise worked for me, In my case, I only had empty `SKIP` files. But then, after creating the `SKIP_IMAGES` files it worked.