RPi-Distro/pi-gen #394: Customize RPi-Distro with node applications

ekaradon, 2020-03-11:
Hello,

First thanks for this repository. It is really useful!
I try to use it in order to store every specific customization of my raspbian in an automated way. However, when I try to use it for node applications it looks like it is not persisted. 
It was thinking that once using `on_chroot << EOF`, everything put within was saved for the latter generated image but it looks like it is not necessary the case.
And then I am a bit annoyed because I try to install things like `nvm` which are setup through shell script, or worse, I try to install npm applications like `pm2` through `npm install pm2 -g` and well I don't see an easy way to repeat the installation without theses.

Here is what my script looks like for now:

```bash
#!/bin/bash -e

# https://github.com/Firerouge/alpine-rtorrentvpn/commit/d2cd7340b74dda4d11b8b447b31e1a1912ee1352
on_chroot << EOF
npm config set unsafe-perm true
EOF

# Patch Issue with ssl certs as per https://github.com/RPi-Distro/pi-gen/issues/271
echo Patching certs...
on_chroot << EOF
c_rehash /etc/ssl/certs
EOF

echo "Installing nvm"
on_chroot << EOF
curl https://raw.githubusercontent.com/creationix/nvm/master/install.sh | bash
EOF

echo "Installing pm2"
on_chroot << EOF
npm install pm2 -g
EOF

on_chroot << EOF
npm config set unsafe-perm false
EOF
```
This script is executed with success, however it does not have any effect when building the image.
Is there a way for me to enforce the build to remember these steps? Or should I just create an rc.local script which will do all this stuff only once the first time the raspberry is launched?

XECDesign, 2020-03-11:
Where are you putting that script? Does it have the +x permission?

XECDesign, 2020-03-11:
Also, that `c_rehash` thing isn't a fix or a workaround. If you're having the issue, described in the linked thread then I would expect all kinds of things to not work.

ekaradon, 2020-03-11:
In stage 3, I have added a 0x-install-node-applications folder with a `00-run.sh` file within which is executable. I see the commands executed by the install (through docker by the way) and it ends gracefully as much as I can see.

XECDesign, 2020-03-11:
No idea why you wouldn't see the changes you expect in the final image. `c_rehash` aside, it looks okay.
You're not skipping any build stages before export? Are you running with `CLEAN=1`?

ekaradon, 2020-03-11:
I have removed the stages 4 and 5 completely. I have also remove the EXPORT_* from stage2. And for the stage3, I have replaced everything with my own scripts. Packages required through apt are installed, customization of systemd works fine. However, I have not used SKIP anywhere as the stage4 and stage5 are erased from the disk.
I am using this command: `PRESERVE_CONTAINER=1 CONTINUE=1 ./pi-gen/build-docker.sh` most of the time. I am going to investigate into this c_rehash thing.

XECDesign, 2020-03-11:
I can't see anything wrong in that case

ekaradon, 2020-03-17:
It seems that the `c_rehash` thing was responsible for this. Once using the right container, I had not anymore this issue. I can close this. Thanks for you help @XECDesign!

closed 2020-03-17 by ekaradon

XECDesign, 2020-03-17:
Glad you've got it working, because I really had no idea why it wouldn't.