RPi-Distro/pi-gen #396: Lite image: /lib   symlink to   /usr/lib  ?

rpier1, 2020-03-25:
Hi,

when comparing lite image obtained with this `pi-gen` script, with the offical lite image (https://downloads.raspberrypi.org/raspbian_lite_latest)

then 
* the one built by this `pi-gen` script has:  
```bash
/bin -> /usr/bin            # symlink???
/lib -> /usr/lib            # symlink???
/sbin -> /usr/sbin          # symlink???
```
while

* the official lite image has:  
```bash
/bin            # no symlink
/lib            # no symlink
/sbin           # no symlink
```

**Questions:**

- Why are there symlinks when using this script?
- And how exactly is the official image (without symlinks) created?

Thanks.

### Details

Running this image-builder to get a **lite** image...

```bash
cat <<EOF        > config
IMG_NAME='Raspbian'
TIMEZONE_DEFAULT=Etc/UTC
ENABLE_SSH=1
STAGE_LIST="stage0 stage1 stage2"
EOF

sed -i 's|FROM debian:buster|FROM i386/debian:buster|' Dockerfile

./build-docker.sh
```

...the result is an image, in which...

(as you can see via `ls -l /`)

```bash
# ls -l /
total 55
lrwxrwxrwx   1 root root     7 Mar 25 12:04 bin -> usr/bin            # symlink???
drwxr-xr-x   3 root root  3072 Jan  1  1970 boot
drwxr-xr-x   7 root root   400 Mar 25 13:16 dev
drwxr-xr-x  79 root root  4096 Mar 25 12:39 etc
drwxr-xr-x   3 root root  4096 Mar 25 12:11 home
lrwxrwxrwx   1 root root     7 Mar 25 12:04 lib -> usr/lib            # symlink???
drwx------   2 root root 16384 Mar 25 12:19 lost+found
drwxr-xr-x   2 root root  4096 Mar 25 12:04 media
drwxr-xr-x   2 root root  4096 Mar 25 12:04 mnt
drwxr-xr-x   3 root root  4096 Mar 25 12:11 opt
dr-xr-xr-x 272 root root     0 Mar 25 13:16 proc
drwx------   2 root root  4096 Mar 25 12:43 root
drwxr-xr-x   3 root root    60 Mar 25 13:16 run
lrwxrwxrwx   1 root root     8 Mar 25 12:04 sbin -> usr/sbin          # symlink???
drwxr-xr-x   2 root root  4096 Mar 25 12:04 srv
dr-xr-xr-x  13 root root     0 Mar 25 11:36 sys
drwxrwxrwt   2 root root    40 Mar 25 13:16 tmp
drwxr-xr-x  10 root root  4096 Mar 25 12:04 usr
drwxr-xr-x  11 root root  4096 Mar 25 12:04 var
```

... in which there are the following 3 symlinks!
```bash
lrwxrwxrwx   1 root root     7 Mar 25 12:04 bin -> usr/bin            # symlink???
lrwxrwxrwx   1 root root     7 Mar 25 12:04 lib -> usr/lib            # symlink???
lrwxrwxrwx   1 root root     8 Mar 25 12:04 sbin -> usr/sbin          # symlink???
```

But the **official lite image** from here

* https://downloads.raspberrypi.org/raspbian_lite_latest
 which currently corresponds to this:
* http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip

does **not have those 3 symlinks**:
```bash
# ls -l
total 67
drwxr-xr-x   2 root root  4096 Feb 13 16:56 bin            # no symlink
drwxr-xr-x   3 root root  3072 Jan  1  1970 boot
drwxr-xr-x   7 root root   400 Mar 25 13:10 dev
drwxr-xr-x  79 root root  4096 Mar 25 11:32 etc
drwxr-xr-x   3 root root  4096 Feb 13 16:55 home
drwxr-xr-x  16 root root  4096 Feb 13 16:58 lib            # no symlink
drwx------   2 root root 16384 Feb 13 17:09 lost+found
drwxr-xr-x   2 root root  4096 Feb 13 16:51 media
drwxr-xr-x   2 root root  4096 Feb 13 16:51 mnt
drwxr-xr-x   3 root root  4096 Feb 13 16:55 opt
dr-xr-xr-x 268 root root     0 Mar 25 13:10 proc
drwx------   2 root root  4096 Mar 25 11:33 root
drwxr-xr-x   3 root root    60 Mar 25 13:10 run
drwxr-xr-x   2 root root  4096 Feb 13 16:58 sbin            # no symlink
drwxr-xr-x   2 root root  4096 Feb 13 16:51 srv
dr-xr-xr-x  13 root root     0 Mar 25 11:36 sys
drwxrwxrwt   2 root root    40 Mar 25 13:10 tmp
drwxr-xr-x  10 root root  4096 Feb 13 16:51 usr
drwxr-xr-x  11 root root  4096 Feb 13 16:51 var
```

### check the image, with mounting

I check the image by mounting it as follows:

(Run this script in a folder holder a `*.img` file)

```bash
#!/usr/bin/env bash

IMG=*.img
MOUNT_IM=/tmp/image_bon

mkdir -p "$MOUNT_IM"

## attach loop device
LOOP=$(sudo losetup -fP --show $IMG) 

sudo mount $FLAGS ${LOOP}p2 $MOUNT_IM
sudo mount $FLAGS ${LOOP}p1 $MOUNT_IM/boot

echo "mounted ${LOOP}p2 on $MOUNT_IM"
echo "mounted ${LOOP}p1 on $MOUNT_IM/boot"

ls -l $MOUNT_IM                 ### here I check the contents of root  /  

## unmount
sudo umount $MOUNT_IM/boot
sudo umount $MOUNT_IM

## detach loop device
sudo losetup -d $LOOP

echo "unmounted and detached"
```

rpier1, 2020-03-25:
Hi, I've found this: https://wiki.debian.org/UsrMerge

It describes how newer systems will have 
"merged `/usr` directories scheme, i.e. the `/{bin,sbin,lib}/` directories becoming symbolic links to `/usr/{bin,sbin,lib}/`."

It says: 
"Merged `/usr` has been the default since debootstrap 1.0.85, reverted in 1.0.87 and re-enabled in 1.0.102."

Under Debian buster, we get:
```bash
qemu-debootstrap --version
# shows 1.0.114
```

As far as I can see... 
this means that the official (lite) image is built from a machine which has a qemu-debootstrap, which still uses the old form (without symlinks).

XECDesign, 2020-04-03:
Thanks for looking into it. The official image should use the merged libraries, so I'll leave it open until that's sorted.

XECDesign, 2020-12-08:
Not quite intentionally, since Ubuntu have updated the version of debootstrap, the last set of images has merged /usr.

closed 2020-12-08 by XECDesign