RPi-Distro/pi-gen #397: Creation of symlink /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3   ?

rpier1, 2020-03-25:
Hi,

where is the following symlink created?
```bash
/lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
```

Command `apt-file search ld-linux.so.3` returns:
```
libc6: /lib/arm-linux-gnueabihf/ld-linux.so.3
libc6-armel-cross: /usr/arm-linux-gnueabi/lib/ld-linux.so.3
libc6-armhf-cross: /usr/arm-linux-gnueabihf/lib/ld-linux.so.3
```
but this does not include `/lib/ld-linux.so.3`.
So it probably happens in some deb's control-file (like `postinst`).

Thanks.

rpier1, 2020-03-25:
Found it!

It happens in the `postinst` file of package `libraspberrypi0` where we find the following

```bash
if [ "$(dpkg --print-architecture)" = armhf ]; then
  ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3 2>/dev/null || true
fi
```

Why is this relevant? Because I removed `libraspberrypi0` from my custom image-build and suddenly got errors such as:
```
-bash: /usr/local/bin/myprog   No such file or directory
```
even though /usr/local/bin/myprog actually exists. The error is due to `file myprog` showing
```
... dynamically linked, interpreter /lib/ld-linux.so.3
```
which is the symlink that I'm missing.

rpier1, 2020-03-25:
Reopen this:

please create the symlink 
```
/lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
```
is this image-builder.

That's safer than having to depend on `libraspberrypi0`. (I don't need that package.)

XECDesign, 2020-04-03:
I wonder why that symlink is needed at all. was 'myprog' cross-compiled or anything?

rpier1, 2020-04-03:
> I wonder why that symlink is needed at all. was 'myprog' cross-compiled or anything?

Yes it was.
Seems other people might be running into this as well: https://github.com/osmc/osmc/issues/339

XECDesign, 2020-04-03:
Then I think your suggestion makes sense.

rpier1, 2020-04-03:
You can put this snippet at the end of `stage0/prerun.sh`

```bash
on_chroot << EOF
ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3 2>/dev/null || true
EOF
```

rafi0101, 2024-07-12:
I have the same issue now 
@XECDesign any plans to do this?
(I use a cross compiled program too)