RPi-Distro/pi-gen #398: Wrong kernel version for custom module

jymboche, 2020-03-26:
I'm trying to compile a wifi driver:
https://github.com/ovhpa/cudy_wu700_ac650

The Makefile uses $(shell uname -a) and the dkms.conf file uses ${kernelver}

uname -a shows:
```Linux adf3269d19a3 4.19.76-linuxkit #1 SMP Thu Oct 17 19:31:58 UTC 2019 armv7l GNU/Linux```

But my /lib/modules folder has:
```
4.19.66+
4.19.66-v7+
```
So my build always fails with:
```
Your kernel headers for kernel 4.19.76-linuxkit cannot be found at
/lib/modules/4.19.76-linuxkit/build or /lib/modules/4.19.76-linuxkit/source.
```
Any guidance on how to solve this would be greatly appreciated! Ive installed `raspberrypi-kernel-headers` in the same stage fyi, also dkms, and ive been trying both regular make install and dkms methods...

XECDesign, 2020-03-30:
Use KVER=blah make. The makefile only uses uname if the version is not set already.

jymboche, 2020-03-30:
Thanks @XECDesign  I used KVER=4.19.66-v7+ on make install as well and it worked. I imagine I'll have to update this if the version of headers installed by raspberrypi-kernel-headers changes? I guess I could check what folders are in /lib/modules dynamically and use that. Before closing this, could you explain the difference between 4.19.66+ and 4.19.66-v7+, and which one I should specify? Thanks!

XECDesign, 2020-04-03:
If you want it to run on every pi model, then you should use each of the versions in /lib/modules/.

I don't remember the exact SoC version v7 is needed for, but 4.19.66+ is for pi 1.

closed 2020-04-03 by jymboche