RPi-Distro/pi-gen #413: Is this the 64-bit raspios pi-gen repository?

davesteele, 2020-06-01:
Will this repository eventually contain the changes used to support generating the 64-bit images for Raspberry Pi OS? If not, does anyone know where the image generator is homed?

I'm confused, because the 32 bit images, which have been renamed, appear to still be part of this chain, but there's no sign of 64-bit support here.

XECDesign, 2020-06-01:
Yes, I'll add an arm64 branch.

davesteele, 2020-06-01:
Thanks. Build tags for the 5/27 build would also be appreciated.

carl1stcallhelpcouk, 2020-06-08:
> Yes, I'll add an arm64 branch.

Any idea when?

luigifreitas, 2020-08-01:
@XECDesign I think this issue could be closed since the `arm64` is now available.

closed 2020-08-25 by XECDesign

fuji246, 2021-11-05:
Is the arm64 branch ready to use?

lurch, 2021-11-05:
Depends what you mean by "ready", but AFAIK the arm64 branch is what's used to create http://downloads.raspberrypi.org/raspios_arm64/images/ (which is still in beta - see https://github.com/raspberrypi/Raspberry-Pi-OS-64bit )