RPi-Distro/pi-gen #414: Unable to get wifi to auto-connect on boot

starbasessd, 2020-06-02:
This used to work:
Build image with minimal changes (read: none)
Add information to config file:
IMG_NAME='Mine'
RELEASE='buster'
LOCALE_DEFAULT='en_US.UTF-8'
KEYBOARD_KEYMAP='us'
KEYBOARD_LAYOUT='English (US)'
TIMEZONE_DEFAULT='US/Eastern'
WPA_ESSID='SSID' 
WPA_PASSWORD='P@$$w0Rd'
WPA_COUNTRY='US'
ENABLE_SSH=1
# APT_PROXY=http://192.168.2.107:3142
# APT_PROXY=http://127.0.0.1:3142
Environment: Debian Buster (10.4) 32 bit
It would build and auto connect to my network, headless without any assistance, allowing me to connect via a statically issued DHCP address.
Now, it refuses saying 
Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.
If I connect keyboard and mouse, or ssh in on ethernet, I can do so,, and it works. 
I have attempted to place the wps_supplicant.conf in /etc/wpa_supplicant folder, during build, no effect.
Is it something that has changed with rfkill, or upstream?
Is there a way to have it enabled by default?

starbasessd, 2020-06-02:
Doing a work-around, was able to force load the wpa_supplicant.conf to {$BOOT_DIR}/ in stage1/00-boot-files/files and force an install in stage1/00-boot-files-00-run.sh, but that exposes the information in the /boot partition prior to first boot.
Since the other options should work, what am I doing wrong?

rnestler, 2020-06-05:
@starbasessd If you want you can test my changes in #416 if they work for you as well :slightly_smiling_face:

starbasessd, 2020-06-05:
I'll check it this weekend.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jessedc, 2020-06-06:
I can confirm the changes in #416 fix the issue :+1: thanks @rnestler

PR #416 (2020-09-18): Only disable wifi if WPA_COUNTRY is not set
Fixes #414
closed 2020-09-18 by XECDesign