RPi-Distro/pi-gen #420: Raspiban vs RPi-OS

132ikl, 2020-06-12:
As mentioned in #413 this repo will soon have a 64 bit branch. Does that mean that both the 32 bit and the 64 bit branches will be renamed to RPi-OS, or will the 32 bit branch remain as Raspbian?

me-at-git-hub, 2020-06-19:
> As mentioned in #413 this repo will soon have a 64 bit branch. Does that mean that both the 32 bit and the 64 bit branches will be renamed to RPi-OS, or will the 32 bit branch remain as Raspbian?

You can see, on the `www.raspberrypi.org/downloads` that the new name for the system image download is no longer Raspbian.  It is now Raspberry Pi OS, which is still 32 bit.  I do not know what the final decision will be on the naming of the 64 bit version of Raspberry Pi OS.  Maybe tacking a 64 on the end?  When the 64 bit version is out,  I would think it would have something in its name to specifically indicate the 64 bit version, so that users, who for whatever the reason want to remain in the 32 bit userland, could clearly distinguish between the 32 bit or the 64 bit Raspberry Pi OS on the downloads page.

lurch, 2020-08-25:
https://github.com/RPi-Distro/pi-gen/commit/78c9e2cf02c312069701dc21f0496a2091b351f4 partially fixes this, but there's still a lot of "Raspbian" mentions in the README.md