RPi-Distro/pi-gen #421: On customizing services

Ryudas, 2020-06-13:
Hello, I have been customizing a version of raspbian for a course, and the last step I would like to configure a service to run on boot. Is there any way to configure a service (with systemctl for example)  from a stage directly?  I have managed to place the service files in their place.

koenvervloesem, 2020-06-13:
I did something like this recently in my `00-run.sh` for a custom Raspbian image created with pi-gen:

```
install -m 644 files/sensor_upload.service "${ROOTFS_DIR}/lib/systemd/system/"

on_chroot << EOF
systemctl enable sensor_upload.service
EOF
```

Ryudas, 2020-06-13:
Ah thank you, I was wondering if systemctl would work on chroot!

closed 2020-06-13 by Ryudas