RPi-Distro/pi-gen #424: Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification.

tihoangyeudau, 2020-06-17:
i have this issues in every utunbu os, raspberry pi os, please help me!

```
Err:1 https://mirror.freedif.org/Raspbian/raspbian buster/main armhf libc-l10n all 2.28-10+rpi1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 113.161.52.138 443]
Err:2 https://mirror.freedif.org/Raspbian/raspbian buster/main armhf locales all 2.28-10+rpi1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 113.161.52.138 443]
W: https://mirror.freedif.org/Raspbian/raspbian/pool/main/g/glibc/libc-l10n_2.28-10+rpi1_all.deb: No system certificates available. Try installing ca-certificates.
W: https://mirror.freedif.org/Raspbian/raspbian/pool/main/g/glibc/locales_2.28-10+rpi1_all.deb: No system certificates available. Try installing ca-certificates.
E: Failed to fetch https://mirror.freedif.org/Raspbian/raspbian/pool/main/g/glibc/libc-l10n_2.28-10+rpi1_all.deb  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 113.161.52.138 443]
E: Failed to fetch https://mirror.freedif.org/Raspbian/raspbian/pool/main/g/glibc/locales_2.28-10+rpi1_all.deb  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 113.161.52.138 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

davesteele, 2020-06-28:
Note https://github.com/RPi-Distro/pi-gen/issues/271

chiptus, 2022-01-06:
#217 suggest using 32bit, but when I use (either on a rpi, or virtual box), I still get these errors (stage-1)

XECDesign, 2022-01-06:
I'm not seeing this.

Does it happen if you build on a Raspberry Pi without Docker?

chiptus, 2022-01-06:
@XECDesign every where I try I get these errors

is there any guide on exactly how to do this?
example:
take rpi4 with X ram, install x os on it with xbits, clone, run this and this command.

XECDesign, 2022-01-06:
Sure, start with an armhf Raspberry Pi OS image on any pi with a large sd card (at least 32GB). I used  Raspberry Pi OS lite on a 4GB pi 4.

Make sure your OS is up to date:
```
sudo apt update
sudo apt full-upgrade -y
```

Install the dependencies and reboot:
```
sudo apt install quilt qemu-user-static debootstrap zerofree zip libarchive-tools file bc qemu-utils kpartx git
sudo reboot
```

Checkout the repo and `cd` into it:
```
git clone https://github.com/RPi-Distro/pi-gen.git
cd pi-gen/
```

Set an image name and kick off a build:
```
echo "IMG_NAME=test" > config
sudo ./build.sh 
```

chiptus, 2022-01-11:
tried the steps you mentioned, still failing

XECDesign, 2022-01-11:
Not much that can be done if it's not a reproducible issue. A bit off that OP's output contains "https" paths, when pi-gen uses "http", so it shouldn't be happening without additional changes.

bhamon, 2022-01-13:
Same issue here with a fresh install on a RPI4. Stage 1 fails on the first apt call because ssl certs are missing in the rootfs.
In the meantime, to bypass this error, copy your system certs to the stage1 directory + add the `ca-certificates` package to ensure proper update:
```
mkdir work/<NAME>/stage1/rootfs/etc/ssl
cp -R /etc/ssl/certs/ work/<NAME>/stage1/rootfs/etc/ssl/certs
echo "ca-certificates" >> stage1/01-sys-tweaks/00-packages
```

XECDesign, 2022-01-13:
If you're changing sources lists to add an https repo (which is the only way I can reproduce OP's issue), you can resolve it by pre-installing `ca-certificates` at the bootstrap stage.

However, apt repos are secure without SSL, so it makes little sense to do so. (https://wiki.debian.org/SecureApt)

```
diff --git a/scripts/common b/scripts/common
index 5b0df12..2c84cbc 100644
--- a/scripts/common
+++ b/scripts/common
@@ -17,6 +17,7 @@ bootstrap(){
        BOOTSTRAP_ARGS+=(--components "main,contrib,non-free")
        BOOTSTRAP_ARGS+=(--keyring "${STAGE_DIR}/files/raspberrypi.gpg")
        BOOTSTRAP_ARGS+=(--exclude=info)
+       BOOTSTRAP_ARGS+=(--include=ca-certificates)
        BOOTSTRAP_ARGS+=("$@")
        printf -v BOOTSTRAP_STR '%q ' "${BOOTSTRAP_ARGS[@]}"
 ```

bhamon, 2022-01-13:
I've used a bare clone of this repo without any modifications.
I double checked the sources.list and sources.list.d in the rootfs and they both contains only http repos.

But still, at stage1 I've got:
```
Réception de :1 http://archive.raspberrypi.org/debian bullseye/main armhf libasound2-data all 1.2.4-1.1+rpt2 [38,7 kB]
Réception de :2 http://archive.raspberrypi.org/debian bullseye/main armhf libasound2 armhf 1.2.4-1.1+rpt2 [304 kB]
Réception de :4 http://archive.raspberrypi.org/debian bullseye/main armhf libatopology2 armhf 1.2.4-1.1+rpt2 [65,9 kB]                
Réception de :5 http://archive.raspberrypi.org/debian bullseye/main armhf alsa-utils armhf 1.2.4-1+rpt1 [1 022 kB]                                                           
Réception de :3 http://distrib-coffee.ipsl.jussieu.fr/pub/linux/raspbian/raspbian bullseye/main armhf pigz armhf 2.6-1 [50,9 kB]              
Réception de :6 http://archive.raspberrypi.org/debian bullseye/main armhf raspi-config all 20220112 [30,2 kB]                                 
Err :7 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf triggerhappy armhf 0.5.0-1.1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :8 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf alsa-topology-conf all 1.2.4-1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :9 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf alsa-ucm-conf all 1.2.4-2
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :10 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgomp1 armhf 10.2.1-6+rpi1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :11 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf libfftw3-single3 armhf 3.3.8-2
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :12 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libsamplerate0 armhf 0.2.1+ds0-1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :13 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf busybox armhf 1:1.30.1-6
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :14 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libklibc armhf 2.0.8-6.1+rpi1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :15 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf klibc-utils armhf 2.0.8-6.1+rpi1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :16 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf initramfs-tools-core all 0.140
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :17 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf linux-base all 4.6
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :18 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf initramfs-tools all 0.140
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :19 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf libnl-3-200 armhf 3.4.0-1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :20 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf libnl-genl-3-200 armhf 3.4.0-1
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
Err :21 https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian bullseye/main armhf iw armhf 5.9-3
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
[... cut ...]
E: Impossible de récupérer https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/w/wireless-regdb/wireless-regdb_2020.04.29-2_all.deb  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Erreur de vérification du certificat. [IP : 193.50.6.155 443]
E: Impossible de récupérer certaines archives, peut-être devrez-vous lancer apt-get update ou essayer avec --fix-missing ?
```

I have no clue where those repos are coming from. Could there be mirror addresses in apt repos that are loaded dynamically?

Anyway, thanks for the tips to bypass this error in the meantime.

XECDesign, 2022-01-13:
@plugwash Is that something that mirrordirector would/should be doing?

plugwash, 2022-01-13:
I think it's the mirror itself doing it.

plugwash@thinkpad:~$ wget http://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb
URL transformed to HTTPS due to an HSTS policy
--2022-01-13 14:32:10--  https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb
Resolving ftp.igh.cnrs.fr (ftp.igh.cnrs.fr)... 193.50.6.155
Connecting to ftp.igh.cnrs.fr (ftp.igh.cnrs.fr)|193.50.6.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 28996 (28K) [application/octet-stream]
Saving to: ‘triggerhappy_0.5.0-1.1_armhf.deb’

triggerhappy_0.5.0- 100%[===================>]  28.32K  --.-KB/s    in 0s      

2022-01-13 14:32:10 (78.2 MB/s) - ‘triggerhappy_0.5.0-1.1_armhf.deb’ saved [28996/28996]

plugwash@thinkpad:~$

plugwash@thinkpad:~$ wget --no-hsts http://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb
--2022-01-13 15:30:07--  http://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb
Resolving ftp.igh.cnrs.fr (ftp.igh.cnrs.fr)... 193.50.6.155
Connecting to ftp.igh.cnrs.fr (ftp.igh.cnrs.fr)|193.50.6.155|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb [following]
--2022-01-13 15:30:07--  https://ftp.igh.cnrs.fr/pub/os/linux/raspbian/raspbian/pool/main/t/triggerhappy/triggerhappy_0.5.0-1.1_armhf.deb
Connecting to ftp.igh.cnrs.fr (ftp.igh.cnrs.fr)|193.50.6.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 28996 (28K) [application/octet-stream]
Saving to: ‘triggerhappy_0.5.0-1.1_armhf.deb.1’

triggerhappy_0.5.0- 100%[===================>]  28.32K  --.-KB/s    in 0.001s  

2022-01-13 15:30:08 (38.2 MB/s) - ‘triggerhappy_0.5.0-1.1_armhf.deb.1’ saved [28996/28996]

plugwash@thinkpad:~$

plugwash, 2022-01-13:
On the one hand my current policy is that mirrors should not be doing this because it breaks older versions of apt, OTOH I expect this behavior will get increasingly common on mirrors over time (as part of the general push towards https that is happening in the world) and I may revise my position on this once stretch drops out of LTS (stretch was the last release where apt needed apt-transport-https to support https retreival).

So I think pi-gen should handle this case.

XECDesign, 2022-01-14:
Ah, many thanks for clearing it up. Didn't think that would be something a mirror could do or apt would allow.

The fix will show up with the next release.

chiptus, 2022-01-16:
> Same issue here with a fresh install on a RPI4. Stage 1 fails on the first apt call because ssl certs are missing in the rootfs (I tried with both `build.sh` and `build-docker.sh`). In the meantime, to bypass this error, copy your system certs to the stage1 directory + add the `ca-certificates` package to ensure proper update:
> 
> ```
> mkdir work/<NAME>/stage1/rootfs/etc/ssl
> cp -R /etc/ssl/certs/ work/<NAME>/stage1/rootfs/etc/ssl/certs
> echo "ca-certificates" >> stage1/01-sys-tweaks/00-packages
> ```

this fixed the issue for me, now seeing other errors (unrelated IMO):

```
[08:35:21] Begin /home/gg/pi-gen/export-image/prerun.sh
du: cannot access '/home/gg/pi-gen/work/tezos/stage4/rootfs': No such file or directory
(standard_in) 1: syntax error
(standard_in) 1: syntax error
Error: The location 272629760 is outside of the device /home/gg/pi-gen/work/tezos/export-image/2022-01-16-tezos.img.
```

commit 40f67ce4 (2022-01-31): Install ca-certificates during bootstrap

Fixes #424
closed 2022-01-31 by XECDesign