RPi-Distro/pi-gen #430: can I use pi-gen to build with an older kernel?

danaronson, 2020-07-22:
For automation we would like to use pi-gen, but have problems with the current buster (4.19.118) kernel.  The previous one (4.19.97) works for us, while we are trying to debug the problem we would like to use pi-gen with the 4.19.97 kernel.  Is there any way to do this?

aniongithub, 2020-07-26:
I need to do this too as I build kernel modules for a specific version of the kernel. Any help is much appreciated!

sensslen, 2020-07-30:
Would #427 address this, as by this request a custom repository containing the installed packages must be generated to be able to reproduce a previous build?

XECDesign, 2024-02-26:
You can use pi-gen to install other kernels, but there's no feature of pi-gen that automates it.

Maybe we will add snapshots to the apt repo and allow pi-gen to build from it, but there are no plans to do that in the near future.

closed 2024-02-26 by XECDesign