RPi-Distro/pi-gen #432: OpenJDK 11 Errors on Debian Buster 32-bit VM

sudeephazra, 2020-07-29:
I am running into errors while building the image. 

This is a fresh VMWare instance running Debian Buster as the Guest OS. 

```
Linux debian-i386 4.19.0-9-686-pae #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) i686 GNU/Linux
```
The error happens on stage5 while trying to install OpenJDK 

Please note that I have not installed anything on this machine except git and the pi-gen dependencies.
 
```
After this operation, 675 MB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf openjdk-11-jre-headless armhf 11.0.7+10-3~deb10u1
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf openjdk-11-jre armhf 11.0.7+10-3~deb10u1
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/o/openjdk-11/openjdk-11-jre-headless_11.0.7+10-3~deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/o/openjdk-11/openjdk-11-jre_11.0.7+10-3~deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

I checked the URL `(openjdk-11-jre-headless_11.0.7+10-3~deb10u1_armhf.deb)` from where the JDK is being attempted to be downloaded and observed that the file itself does not exist. I see another alternate file though `(openjdk-11-jre-headless_11.0.7+10-1~deb10u1_armhf.deb)` and this seems to be legit.

How can I resolve this and tell apt-get to use the correct file?

XECDesign, 2020-07-29:
Debian have recently pushed an update to OpenJDK. I think the build started before it was pulled into Raspbian, but by the time it was trying to install OpenJDK, the version was no longer there. A clean build should fix it.

sudeephazra, 2020-07-30:
Tried a clean build as well but ran into the same issue. Moved the build process to the Rpi and it seemed to work. The image has been generated successfully on the RPi.

closed 2020-07-30 by sudeephazra