RPi-Distro/pi-gen #433: Include instructions for using mirrors in Readme

popnt, 2020-08-03:
Given how popular the main repository is, performance is often impacted whereby transfers are slow, and failures frequent.

While I am no expert on Pi-gen my understanding is that even a single missing archive will result in having to start build.sh over again from scratch, with CLEAN=1, and having to wait typically hours again to transfer everything from raspberrypi.org

Ideally it would be better to include a parameter directly in "config" so that using a mirror does not mean having to edit files individually, but would likely mean updating main pi-gen code, whereby updating the Readme should be trivial.

The Readme should include the files that need updating, and as a convenience a link to the official list of mirrors https://raspbian.org/RaspbianMirrors

and I believe the individual files are the following:
- /Pi-gen/stage0/prerun.sh
- /Pi-gen/stage0/00-configure-apt/files/sources.list
- /Pi-gen/stage0/00-configure-apt/files/raspi.list

Lastly it should be noted that the default repository raspbian.raspberrypi.org does not support HTTPS however archive.raspberrypi.org does, yet raspi.list and indicated above uses HTTP anyway: Some might prefer HTTPS and updating the Readme with the above would highlight those details as configurable.

cdupont, 2021-11-30:
I currently have this problem. The address http://raspbian.raspberrypi.org/raspbian is failing or extremely slow.
My builds are failing with:
```
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Checking component main on http://raspbian.raspberrypi.org/raspbian...
I: Retrieving adduser 3.118
I: Validating adduser 3.118
W: Retrying failed download of http://raspbian.raspberrypi.org/raspbian/pool/main/a/adduser/adduser_3.118_all.deb
```
Any suggestion?

XECDesign, 2022-01-11:
pi-gen isn't really set up with mirrors in mind, so I'm not sure much can be done on this end.

We use `apt-cacher-ng` and the `APT_PROXY` config option to ensure builds don't fail. It can also map problematic URLs to known good ones. https://www.unix-ag.uni-kl.de/~bloch/acng/html/index.html

Alternatively, you could add something to your local fork to modify sources lists to suit your use case.