RPi-Distro/pi-gen #434: REQUEST: ability to create x86 images

puccaso, 2020-08-03:
the raspberry pi os has proved very successful for me on my desktop(s)
i'd love the ability to create my own images with pi-gen,
it also solves a problem, as after you burn the rpdx86iso - you can't edit any contents on the drive, 
but if we could make our own, it would probably be much more successful than unpacking the iso, editing, then re-isoing the directory. 
could this be an option?

puc

mdellabitta, 2021-05-16:
I've been doing some dev work meant to run on a pi from my laptop. Would be nice to be able to test them in Vagrant...

sethleedy, 2023-01-16:
A single tool to create all arch types would be nice.
pi-gen --arch-list x86_64 arm64