RPi-Distro/pi-gen #438: No space left on device on node_modules

jaumard, 2020-08-24:
I'm getting errors with "No space left on device" when generating the image.

How can I fix that ? Should I change `ROOT_MARGIN` in prerun.sh and run `CONTINUE=1 ./build-docker.sh` ? 

Or is there another(faster) way to fix this ?

jaumard, 2020-08-25:
I've increase the size and now I'm not getting this error but instead the process look freezed after:

```
[10:46:37] Begin /pi-gen/export-image/prerun.sh
IMAGE SIZE: 4219469824

/boot: offset 4194304, length 268435456
/: offset 272629760, length 3946840064
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mke2fs 1.44.5 (15-Dec-2018)
mount: /dev/loop3 mounted on /pi-gen/work/2020-08-25-mylisabox-buster/export-image/rootfs.
mount: /dev/loop2 mounted on /pi-gen/work/2020-08-25-mylisabox-buster/export-image/rootfs/boot.
```
The docker show no more activity once this is reach
[image]

Is it because the image size is more than 4G ? How can I unlock this ? (I'm on mac os X)

starbasessd, 2020-09-19:
I'm doing it in a Debian VM (not docker) I downloaded the latest setup, had 48GB free space and it I got this error. I disabled Stage5, and all EXPORT_NOOBS. and it works.