RPi-Distro/pi-gen #443: Creating folders in the raspbian distribution itself

Ryudas, 2020-09-04:
hello I was trying to create folders in the generated raspbian to put some scripts by default. It seems this does not work:
```bash
#!/bin/bash -e

# place scripts services depend on
install -v -d                          "${ROOTFS_DIR}/etc/systemd/system/service_scripts/python"
install -v -m 755 files/ip.py	       "${ROOTFS_DIR}/etc/systemd/system/service_scripts/python/"
``` 

Is there a way for me to do this?

XECDesign, 2020-09-04:
Is there an error when that runs?

I'm guessing that `${ROOTFS_DIR}/etc/systemd/system/service_scripts` doesn't exist, so you need to create it the same way first.

Ryudas, 2020-09-04:
> Is there an error when that runs?
> 
> I'm guessing that `${ROOTFS_DIR}/etc/systemd/system/service_scripts` doesn't exist, so you need to create it the same way first.

I will try to find it, however, does  install -v -d   not create the directory, if it does not exist? i had thought that was default behaviour

XECDesign, 2020-09-04:
You're right, that should work as far as I can tell.

Is this a clean build or are you reusing built stages?

Ryudas, 2020-09-04:
I've gotten the same result either from a  clean build (from the build.sh)  or the docker build.  in both cases the folders are not in the final place and the files supposed to be there do not seem to have the permissions.

Ryudas, 2020-09-04:
> I've gotten the same result either from a clean build (from the build.sh) or the docker build. in both cases the folders are not in the final place and the files supposed to be there do not seem to have the permissions.

here are the outputs of logs:
IM INSTALLING FOLDERS
install: creating directory '/pi-gen/work/2020-09-04-DCDbian/stage2/rootfs/etc/systemd/system/dhcpcd.service.d'
'files/wait.conf' -> '/pi-gen/work/2020-09-04-DCDbian/stage2/rootfs/etc/systemd/system/dhcpcd.service.d/wait.conf'
'files/wpa_supplicant.conf' -> '/pi-gen/work/2020-09-04-DCDbian/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
'files/eduroam.sh' -> '/pi-gen/work/2020-09-04-DCDbian/stage2/rootfs/etc/systemd/system/service_scripts/eduroam.sh'
'files/eduroam.service' -> '/pi-gen/work/2020-09-04-DCDbian/stage2/rootfs/etc/systemd/system/eduroam.service'
[17:31:11] End /pi-gen/stage2/02-net-tweaks/01-run.sh
[17:31:11] End /pi-gen/stage2/02-net-tweaks

here is what i put in the beginning of that script: 
```bash

install -v -d                             "${ROOTFS_DIR}/etc/systemd/system/dhcpcd.service.d"
install -v -m 644 files/wait.conf         "${ROOTFS_DIR}/etc/systemd/system/dhcpcd.service.d/"

install -v -d                             "${ROOTFS_DIR}/etc/wpa_supplicant"
install -v -m 600 files/wpa_supplicant.conf      "${ROOTFS_DIR}/etc/wpa_supplicant/"

# place scripts services depend on
mkdir -p "${ROOTFS_DIR}/etc/systemd/system/service_scripts"
install -v -d                             "${ROOTFS_DIR}/etc/systemd/system/service_scripts"
install -v -m 755 files/eduroam.sh	       "${ROOTFS_DIR}/etc/systemd/system/service_scripts/"	

# place eduroam services
install -v -m 644 files/eduroam.service       "${ROOTFS_DIR}/etc/systemd/system/"
```

Ryudas, 2020-09-04:
As you can see i only see the log for the first one_ ?

XECDesign, 2020-09-04:
Judging by the logs, it looks like it's working just fine. Are you 100% sure something else isn't happening between the time stage 2 runs and wherever you're checking whether the files exist?

Ryudas, 2020-09-04:
these are the logs after i added the mkdir instruction.  the logs without it are like so:
[image]

I have only changed stage 2 and stage 5, not sure what I could have done in between besides using on_chroot for enabling some services .

Ryudas, 2020-09-08:
Can confirm using mkdir works normally but install for some reason does not create the fokders using the script

closed 2020-09-08 by Ryudas

XECDesign, 2020-09-08:
How strange... we clearly create directories earlier in the same file using install. And your output earlier showed the files being copied... if the directory wasn't there, it wouldn't have worked.

Ryudas, 2020-09-08:
Yes indeed which is why initially i was confused. indeed the later logs do show that the folder and file is installed, but later when i check the image the new folders (without using mkdir) are not there (and neither the files that should be inside) . with mkdir its immediately solved. Not sure what is the problem.