RPi-Distro/pi-gen #446: Failure while configuring base packages

Ryudas, 2020-09-15:
So Building on a debian server, with docker build, the images were being generated before, but now they fail witht ht e following log:

```bash
I: Configuring gpg-wks-client...

I: Configuring man-db...

I: Configuring gnupg...

I: Configuring raspbian-archive-keyring...

I: Configuring systemd...

W: Failure while configuring base packages.  This will be re-attempted up to five times.

W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)

I: Configuring raspbian-archive-keyring...

W: Failure while configuring base packages.  This will be re-attempted up to five times.

W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)

W: Failure while configuring base packages.  This will be re-attempted up to five times.

W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)

W: Failure while configuring base packages.  This will be re-attempted up to five times.

```
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)

W: Failure while configuring base packages.  This will be re-attempted up to five times.

W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)

XECDesign, 2020-09-16:
Our nightlies still seem to be building, so I'm not sure that anything has changed that just broke everything for everybody.

Is there any more information you can get? Maybe from debootstrap.log or trying to run debootstrap manually with debug/verbose options turned on from within the appropriate docker container?

Ryudas, 2020-09-16:
Under more research We found the issue seems to be from running multiple containers concurrently. Right now weŕe enforcing a strictly sequential generation. Further probing  eventually lead us to a modle not found fatal error with mobprobe and  a linux kernel module, in our case 4.19.0-10-686-pae

Ryudas, 2020-09-18:
Ensuring the containers run sequentially fixes this erro for now, so I'll close it barring other solutions

closed 2020-09-18 by Ryudas