RPi-Distro/pi-gen #453: Changed needed in wifi configuration. Unbroadcast SSIDs

MichaelTiernan, 2020-09-29:
Some of us have wifi that isn't broadcast so we need the phrase `ssid_scan=1` in the wpa_supplicant file.

starbasessd, 2020-09-30:
I don't know if this is an 'approved way' but I made a modification to pi-gen/stage2/02-net-tweaks/01-run.sh:
install -v -d "${ROOTFS_DIR}/etc/wpa_supplicant/wpa_supplicant.conf"
install -v -m 600 files/wpa_suplicant.conf  "${ROOTFS_DIR}/etc/wpa_supplicant/wpa_supplicant.conf"
I edited/added my wpa_supplicant.conf in pi-gen/stage2/02-net-tweaks/files
This has worked for me for the past month or 2.
I gave up on trying to get the other fixes to work.

XECDesign, 2020-10-01:
That's 100% the intended way to make such changes.