RPi-Distro/pi-gen #454: Issue 338 & 294 back? rmdir: failed to remove '/<path>/pi-gen/work/2020-10-03-test/stage0/rootfs/debootstrap': Directory not empty

starbasessd, 2020-10-03:
Situation: 
New setup. VBox VM 6.1.14. Debian Bullseye 32 bit install 4GB RAM & 50GB HD 
full update & upgrade. (worked through mid-Sept in another VM)
fresh git clone of sources.
Installed all dependencies per main page.
config has IMG_NAME='test'
Checked scripts/common, looks the same as the fix posted.
shows Running command: chroot /home/kevin/pi-gen/work/2020-10-03-test/stage0/rootfs
Installing core packages...
a bunch of unpacking & configuring then 
unpacking the base system
then 
W: Failure trying to run:  dpkg --force-overwrite --force-confold --skip-same-version --install /var/cache/apt/archives/adduser_3.118_all.deb /var/cache/apt/archives/dmsetup_2%3a1.02.155-3_armhf.deb /var/cache/apt/archives/libapparmor1_2.13.2-10_armhf.deb /var/cache/apt/archives/libargon2-1_0~20171227-0.2_armhf.deb /var/cache/apt/archives/libcap2_1%3a2.25-2_armhf.deb /var/cache/apt/archives/libcryptsetup12_2%3a2.1.0-5+deb10u2_armhf.deb /var/cache/apt/archives/libdevmapper1.02.1_2%3a1.02.155-3_armhf.deb /var/cache/apt/archives/libffi6_3.2.1-9_armhf.deb /var/cache/apt/archives/libgmp10_2%3a6.1.2+dfsg-4_armhf.deb /var/cache/apt/archives/libgnutls30_3.6.7-4+deb10u5_armhf.deb /var/cache/apt/archives/libhogweed4_3.4.1-1_armhf.deb /var/cache/apt/archives/libidn11_1.33-2.2_armhf.deb /var/cache/apt/archives/libidn2-0_2.0.5-1+deb10u1_armhf.deb /var/cache/apt/archives/libip4tc0_1.8.2-4_armhf.deb /var/cache/apt/archives/libjson-c3_0.12.1+ds-2+deb10u1_armhf.deb /var/cache/apt/archives/libkmod2_26-1_armhf.deb /var/cache/apt/archives/libnettle6_3.4.1-1_armhf.deb /var/cache/apt/archives/libp11-kit0_0.23.15-2_armhf.deb /var/cache/apt/archives/libseccomp2_2.3.3-4_armhf.deb /var/cache/apt/archives/libssl1.1_1.1.1d-0+deb10u3_armhf.deb /var/cache/apt/archives/libtasn1-6_4.13-3_armhf.deb /var/cache/apt/archives/libunistring2_0.9.10-1_armhf.deb /var/cache/apt/archives/systemd_241-7~deb10u4+rpi1_armhf.deb
W: See //debootstrap/debootstrap.log for details (possibly the package systemd is at fault)
rmdir: failed to remove '/home/kevin/pi-gen/work/2020-10-03-test/stage0/rootfs/debootstrap': Directory not empty
debootstrap.log shows 
Setting up systemd (241-7~deb10u4+rpi1) ...
Allocating guest commpage: Cannot allocate memory
Allocating guest commpage: Cannot allocate memory
Allocating guest commpage: Cannot allocate memory
Allocating guest commpage: Cannot allocate memory
dpkg: error processing package systemd (--install):
 installed systemd package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.28-10+rpi1) ...
Errors were encountered while processing:
 systemd
free shows 75M used, 2.8G free, df -h shows 42GB free.
Thoughts? Currently running on older VM, and running copy of older VM pi-gen tree on new VM. Will update.

starbasessd, 2020-10-03:
Update 1:
copied build tree of pi-gen to the new VM, same issue.
Old VM runs fine with the old pi-gen.

starbasessd, 2020-12-31:
Closing after no response in 90 days

closed 2020-12-31 by starbasessd

L3P3, 2021-04-26:
I get the same error when trying to run an electron-based application in qemu-user-armhf:
```
Allocating guest commpage: Cannot allocate memory
```