RPi-Distro/pi-gen #460: Replace apt-key due to its deprecation

MichaIng, 2020-10-31:
`apt-key` has been deprecated ans scheduled for removal with Q2/2022: https://metadata.ftp-master.debian.org/changelogs//main/a/apt/apt_2.1.11_changelog

It's still some time but I suggest to update `pi-gen` to not use it anymore, as currently done in stage 0 to add the key for archive.raspberrypi.org: https://github.com/RPi-Distro/pi-gen/blob/master/stage0/00-configure-apt/00-run.sh

The `raspberrypi-archive-keyring` package from this repo ships the key as `/etc/apt/trusted.gpg.d/raspberrypi.gpg` which is probably a reasonable solution anyway to have the key not in `/etc/apt/trusted.gpg` but transparently as part of a DEB package, pre-installed on Raspberry Pi OS images. Although I'd personally prefer `/etc/apt/trusted.gpg.d/raspberrypi-archive-keyring.gpg` for consistency with Debian and Raspbian keyring packages and names and even the one shipped with `raspberrypi-archive-keyring` in `/usr/share/keyrings/raspberrypi-archive-keyring.gpg`.

Not sure how this chicken-egg issue is solved usually, probably the key could be stored in APT-compatible format here in the repo, basically replacing https://github.com/RPi-Distro/pi-gen/blob/master/stage0/00-configure-apt/files/raspberrypi.gpg.key and being simply copied to `/etc/apt/trusted.gpg.d`, then replaced by the DEB package afterwards?
Alternatively the following would work as a single-line change:
```
gpg --dearmor -o /etc/apt/trusted.gpg.d/raspberrypi.gpg --yes < files/raspberrypi.gpg.key
```
_____________
It's not about `pi-gen` but related: The `raspbian-archive-keyring` package requires an update as well. It currently ships the key in APT-compatible format in `/usr/share/keyrings/raspbian-archive-keyring.gpg` but adds it via `apt-key` in `postinst` script. For comparison, `debian-archive-keyring` ships the key in both locations, `/usr/share/keyrings/debian-archive-keyring.gpg` and `/etc/apt/trusted.gpg.d/debian-archive-keyring.gpg` to be effective without any `postinst` script. I'd personally prefer a symlink or `cp` (in `postinst`), but either works as a simple solution to replace `apt-key`. Just not sure where to report/suggest Raspbian-related issues?

XECDesign, 2020-11-02:
Thanks for the detailed report. I can take a look at the pios keyring package at some point. Issues with packages that come from raspbian.org can be reported here - https://bugs.launchpad.net/raspbian

MichaIng, 2020-11-03:
On Raspbian bug tracker there was an old related report already, not reported for the same reason but same aim, where I added info and reference: https://bugs.launchpad.net/raspbian/+bug/1727874
Will see if a bug report from 2017 still gets attention at all 😄.

XECDesign, 2022-02-23:
Should be resolved since 231487d9

closed 2022-02-23 by XECDesign

beta-tester, 2024-11-13:
still an issue in 
[2024-10-22-raspios-bookworm-armhf-lite.img.xz](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2024-10-28/2024-10-22-raspios-bookworm-armhf-lite.img.xz)