RPi-Distro/pi-gen #464: Install custom Debian packages including dependencies

b-s101, 2020-11-09:
Hi,

I need to install a custom debian package during the build process. Its dependencies should be installed from the Debian repositories. Is there any recommended way to install custom packages?

Thanks

koenvervloesem, 2020-11-09:
I did this recently like this to install Node-RED from its official installation instructions:

* Create a directory `04-install-nodered` in `stage2`.
* In `00-packages` put its dependencies (`git npm` in this example). They will be installed by pi-gen.
* Create a file `00-run.sh` with the installation script.

The latter looks like this:

```
#!/bin/bash -e

# Install Node-RED
on_chroot <<EOF
su - pi -c "bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered) --confirm-install --confirm-pi"
EOF
```
So you just run the installation script in the chroot. The dependencies are already installed when the script runs because `00-packages` comes before `00-run.sh` alphabetically.

I hope this helps as a guideline for your own custom package.

b-s101, 2020-11-09:
Thanks, it worked. The only difference is that the package I need to install is not available on a webserver. So I put the package to 04-install-mypackage/files/package.deb, copied it to tmp/ inside the chroot env and removed it from there after installation.
Is there any way to access the files-directory from inside the chroot environment?

koenvervloesem, 2020-11-09:
No the chroot environment doesn't see anything outside the root file system that you're building with pi-gen. So the way you have done this is fine.

koenvervloesem, 2020-11-09:
Now that I think of it: there's an even cleaner way to do this.

* Copy your `package.deb` from `files` to `/var/cache/apt/archives/` in the chroot.
* Install the package in the chroot.

Then pi-gen should clean the package because of the `apt-get clean` command in [export-image/01-set-sources/01-run.sh](https://github.com/RPi-Distro/pi-gen/blob/master/export-image/01-set-sources/01-run.sh).

b-s101, 2020-11-09:
Right. That would make it even cleaner. 
Thanks!

RiverHeart, 2022-11-17:
Small note that `dpkg` -i won't be able to resolve any dependencies via Apt when installing the deb package. You can, however, invoke `apt install` against the deb file on disk and Apt will resolve the dependencies. This means you don't need to manage those dependencies via a packages file which is nice when things change.

XECDesign, 2022-11-17:
`apt install ./file.deb` might (note the `./`, which makes sure apt treats it as a local file rather than a package name)