RPi-Distro/pi-gen #470: Unable to install docker app images as part of a custom build stage

ivelin, 2020-11-18:
While building a custom Raspberry OS image, I would like to add a stage that installs a few docker images which will run when the image is booted on a raspberry pi device. I am able to install docker and docker-compose, however I am unable to install images, because docker fails to start. It logs the following error message:

```
failed to start daemon: Error initializing network controller: error obtaining controller instance: 
failed to create NAT chain DOCKER: iptables failed: iptables -t nat -N DOCKER: 
iptables/1.8.2 Failed to initialize nft: Protocol family not supported
```

[Here is the full context](https://github.com/ambianic/ambianic-rpi-image/runs/1415143806#step:9:5171) of the github action build log and error message.

cdupont, 2021-09-13:
Hello, have you been able to solve this problem?
To install docker itself, I suppose you added it in `00-packages` file?

ivelin, 2021-09-13:
Not yet, thank you for checking. Docker engine installs but I am unable to make it pull the latest docker images during the custom RPI OS image build. The workaround is to [pull the docker images on boot ](https://github.com/ambianic/ambianic-quickstart/blob/52163068f4a66f90971bee5172b943b887e494de/scripts/docker-compose-start.sh#L7) ( when the custom RPI OS boots for the first time). This is not ideal because it makes the device unusable for the first 5-10 minutes.

cdupont, 2021-09-13:
Yeah pulling images on boot is not ideal.
What about making a `docker save` on the host, move the tar file, and `docker load` at startup?

cdupont, 2021-09-13:
https://github.com/RPi-Distro/pi-gen/issues/293

ivelin, 2021-09-13:
Thank you for referencing #293 . Good to know that save/load is a another workaround. Any idea what the root cause is for the error on docker pull during RPI OS build?

cdupont, 2021-09-17:
I managed to reproduce your bug:
```
Error starting daemon: Error initializing network controller: error obtaining controller instance: failed to create NAT chain DOCKER: iptables failed: iptables -t nat -N DOCKER : iptables/1.8.2 Failed to initialize nft: Protocol family not supported
```

I got the error while trying to start docker deamon on the chroot:
```
on_chroot << EOF
cgroupfs-mount
/usr/sbin/dockerd
EOF
```
Is it what you did? Is it the correct approach?
My goal is to simply `docker pull` a docker image, so that it is ready for the ISO first boot.

cdupont, 2021-09-17:
I followed the advice from here: https://superuser.com/questions/1480986/iptables-1-8-2-failed-to-initialize-nft-protocol-not-supported
```
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
```
But then some other problem appeared (wrong modules).

There doesn't seem to be any hope.

jedahan, 2022-12-29:
Also interested in running docker commands during the build of the image

beniroquai, 2024-09-20:
Is there any update on this? A simple `docker pull` would already be great. :)