RPi-Distro/pi-gen #471: Feature request: Support F2FS filesystem for root partition

jimis, 2020-11-22:
I guess this is more appropriate project to report this feature request. Copying verbatim from https://github.com/raspberrypi/noobs/issues/596 :

[F2FS](https://en.wikipedia.org/wiki/F2FS) has been designed on the basis of log-structured filesystems, with the goal of performing optimally on managed flash media like SD cards, minimizing writes and extending their lifetime. It has seen some usage on Android phones, and it's constantly being improved with features that Raspberry Pi would very much like to have, like online compression.

Currently Raspberry Pi OS root partition can be converted to F2FS using a time-consuming process:
* Mount SD card with Linux PC
* Back up root partition
* Make F2FS filesystem on the SD card
* Restore backup to the new partition
* Update `fstab` and boot command line on the SD card

This is a dangerous and write-intensive process. In order to avoid it, it would be nice if installation could be done directly on a F2FS root partition.

XECDesign, 2020-11-23:
Just waiting for some evidence that F2FS fixes more issues than it causes. At the very least, right now runs fsck on every single boot and I'm not sure that it can recover if it finds an issue, since it's running on a mounted partition. I suspect it will need initramfs to work properly.

https://www.raspberrypi.org/forums/viewtopic.php?t=273490

rekcodocker, 2021-10-03:
If you want evidence, it would be nice to create an image which runs on F2FS. Call it 'experimental' and collect evidence. 
Right now there are 3 options if someone wants to use F2FS: 
- there is an older Rasbian image (I believe 2020);
- there is a conversion script ( cv2f2fs ).  Very nice but cumbersome. Requires an extra USB disk. Does not support compression. 
 Not for the average user.
- DIY - also not for the average user. 

It'd be nice to add a smoother more accessable option.  No need to throw out EXT4; just to gather evidence.

XECDesign, 2021-10-03:
There'a already a link to an experimental image.

rekcodocker, 2021-10-03:
Indeed there is!
Apologies for my thankless attitude, however, that particular image is over a year and a half old now.

trosenkranz, 2022-10-13:
Hello everybody,

Is there any update on this issue?

Thanks Tobias

TipsyTheCat, 2024-05-31:
It's been 4 years, F2FS has had many improvements and i think it's time this gets more attention