RPi-Distro/pi-gen #476: Example custom stage

mikysal78, 2020-12-02:
if i need to compile and install a software what can i do?
thanks

XECDesign, 2020-12-02:
That question is a bit too open-ended. What have you tried? What was the result? Is there anything in particular that's unclear? The existing stages show how to run commands inside and outside of the image environment and how to move files around.

mikysal78, 2020-12-02:
example: when I generate image I want to compile and configure olsrd2 without modifying the default stages

Claudenw, 2020-12-31:
I did a quick scan through the olsdr2 pages and it looks like the standard procedure there is to build a debian package.  If that is the case then you could:

 1. create a local apt repository (I have an example in https://github.com/Claudenw/pi-gen-485 (which is for a bug report but the apt part works) and place the debian package there.
 2. add an entry for your private repository (see last line of `setup.sh` in above project)
 3. add steps in the appropriate stage to register the apt in the build process (see `pi-gen-overlay/stage3/02-configure-apt`  in the above project)
 4. add steps to actually install the module using a nn-Packages file (see `pi-gen/stage3/00-install-packages`  for an example)
 5. run the pi-gen build.

If the olsdr2 requires compilation for a kernel you could look at using dkms, but I have not been able to get it to work (thus the bug report).

mikysal78, 2021-02-13:
> I did a quick scan through the olsdr2 pages and it looks like the standard procedure there is to build a debian package. If that is the case then you could:

This is first compile olsrd2, create repo etc..after compiling image with pi-gen.
i wanted the compilation of olsrd2 and install to take place during image creation with pi-gen

capiman, 2021-02-26:
Hello,

> i wanted the compilation of olsrd2 and install to take place during image creation with pi-gen

I am currently searching for the same.
But I am not yet sure, if pi-gen just collects pre-compiled parts/packages or also supports compilation during image creation?

Best regards,
Marie

capiman, 2021-02-26:
Can I just add something like

git clone https://github.com/xyz/abc
cd abc
./bootstrap
./configure
make
make install
cd ..

to one of the files?

closed 2026-10-29 by mikysal78