RPi-Distro/pi-gen #478: curl not functional in newly built image

marcone, 2020-12-10:
I built a 'lite' image using pi-gen (so stages 0-2, but not 3 and further), but while the resulting image boots just fine, `curl` doesn't work on it:
```
$ curl https:///www.example.com
curl: (60) SSL certificate problem: self signed certificate in certificate chain
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```
This can be fixed by running `/usr/sbin/update-ca-certificates --fresh` (or alternatively `apt remove ca-certificates` followed by `apt install ca-certificates`), which would seem to imply that when the package is installed by pi-gen, it is missing this step.

jimmy1248, 2021-04-16:
I can confirm that I am having the same issue. The fix works when you boot the image. I am building the lite image. 

The pre-built version here https://www.raspberrypi.org/software/operating-systems/ does work with curl out of the box.

The fix doesn't work when you want to use curl during the build.

jketterl, 2021-04-16:
This may be related to #271.

XECDesign, 2021-09-08:
Duplicate of https://github.com/RPi-Distro/pi-gen/issues/271

closed 2021-09-08 by XECDesign