RPi-Distro/pi-gen #481: raspios_lite_arm64 images are not being updated

guysoft, 2020-12-15:
Hey,
I maintain sevral distros, mainly OctoPi, and it seems like there issues with the raspios_lite_arm64 images.
However, it seems like they are not being updated neither. The latest on is ``2020-08-24`` as see here:
https://downloads.raspberrypi.org/raspios_lite_arm64/images/?C=M;O=A

While raspios_lite_armhf has been updated this month to ``2020-12-04`` as see here: https://downloads.raspberrypi.org/raspios_lite_armhf/images/

This is halting our release candidate here:
https://github.com/guysoft/OctoPi/issues/692

People are getting issues with arm64 that might be solved if the firmware is updated. For example:
https://github.com/guysoft/OctoPi/issues/697

XECDesign, 2020-12-15:
The 64bit builds are for beta testing and should not be relied on. The next one will switch over to 5.10 and use KMS rather than FKMS, so there's a bit of a delay.

guysoft, 2020-12-15:
Is there another issue or place worth following for this kind of update?

XECDesign, 2020-12-15:
I expect there will be a thread on https://www.raspberrypi.org/forums/viewforum.php?f=117 once the next update is released.

guysoft, 2022-06-20:
> The 64bit builds are for beta testing and should not be relied on.

@XECDesign Hey, this came out on Feb: https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/

Does this mean it is out of beta? :)

XECDesign, 2022-06-22:
Yup, and they should be getting updated at the same time as all the other images.

guysoft, 2022-06-22:
Ok, I guess this issue can be closed then

closed 2022-06-22 by XECDesign