RPi-Distro/pi-gen #482: losetup fails with docker-in-docker

bamartin125, 2020-12-15:
The scenario is this:
* Run pi-gen build inside docker-in-docker served from a dockerized CI build agent (`teamcity-agent`, etc.)
* Host `/dev` is *not* volume mounted to container `/dev` for isolation reasons
* The build agent and the pi-gen docker image are `--privileged` (necessary for `losetup`)
* All loop devices are currently being used, but new ones may be created

With this scenario, a call to `losetup ... -f ... <file>` instructs the kernel to create a new loop device which is subsequently mapped to the hosts `/dev` directory, but is not automatically populated in the container's `/dev` directory. This ultimately causes that previous `losetup` call to fail because `losetup` cannot map the `<file>` to the loop device because the loop device cannot be found by the container. If the container is `--privileged`, then the container can in fact make it's own loop device node such that a subsequent call to `losetup` can properly map the `<file>`.

Related issue: https://github.com/moby/moby/issues/27886

Per that issue, some have "solved" the issue by one of the following:
* volume mount host `/dev` to container `/dev`
* use the host's docker instance (volume mount `/var/run/docker.sock`) rather than an isolated docker instance (docker-in-docker)
* require/use a FUSE filesystem in place of `losetup` to access the image files as required by the build
* have the container (with `--privileged` flag) create it's own loop device handles with `mknod`

In my case, I would prefer not to use either of the first two options. I also believe that the pi-gen project would not like to break away from `losetup` (and require/use a FUSE solution) without enough testing.

The proposal then would be to enhance both `export-noobs/prerun.sh` and `export-image/prerun.sh` to ensure that the next loop device is available before `losetup ... -f ... <file>` is called. This would greatly enhance the chances of `losetup` being successful.

I plan to submit a PR for this issue shortly.

dfloer, 2021-01-05:
Just wanted to add that I have the same issue as well (using the docker setup on an x86 box so I can create the ARM images from there as part of an automated build process), and the code in the linked PR did indeed allow pi-gen to generate an image instead of failing.

SerhiyMakarenko, 2023-08-12:
+1. I ended up with a solution from the related issue [moby/moby/issues/27886](https://github.com/moby/moby/issues/27886):
```
	PARTITIONS=$(lsblk --raw --output "MAJ:MIN" --noheadings ${LOOP_DEV} | tail -n +2)
	COUNTER=1
	for i in $PARTITIONS; do
		echo "Creating node file for partition $i..."
	    MAJ=$(echo $i | cut -d: -f1)
	    MIN=$(echo $i | cut -d: -f2)
	    if [ ! -e "${LOOP_DEV}p${COUNTER}" ]; then mknod ${LOOP_DEV}p${COUNTER} b $MAJ $MIN; fi
	    COUNTER=$((COUNTER + 1))
	done
```
Included into `pi-gen/export-image/prerun.sh` and it also works.

However, would be great if pull request that fix this issue can be merged.

closed 2023-08-17 by XECDesign

acinonyx, 2024-01-03:
This issue has not been completely fixed with #483 . The partition devices may not be created. #741 fixes this issue by introducing a similar function `ensure_loopdev_partitions`.

PR #741 (2024-01-04): Ensure loop device partition nodes are created
Although the loop block device is created before attaching the image to it, the devices for the partition that the image contains are still not created. This patch creates those devices as well, when they are not already available.

Fixes #482
commit 1377e903 (2024-01-04): Ensure loop device partition nodes are created (#741)

Although the loop block device is created before attaching the image
to it, the devices for the partition that the image contains are still
not created. This patch creates those devices as well, when they are
not already available.

Fixes #482
commit 92791ed7 (2024-02-26): Ensure loop device partition nodes are created (#741)

Although the loop block device is created before attaching the image
to it, the devices for the partition that the image contains are still
not created. This patch creates those devices as well, when they are
not already available.

Fixes #482