RPi-Distro/pi-gen #485: DKMS is not supported on pi-gen

Claudenw, 2020-12-24:
I am trying to install a DKMS managed package as part of pi-gen build using the docker build on Ubuntu 18.04.5 LTS.

However, the DKMS build seems to detect the underlying (Ubuntu) kernel version not the version being installed by pi-gen.  I have documented the issue on [Raspberry Pi Stack Exchange](https://raspberrypi.stackexchange.com/questions/119494/using-dkms-in-pi-gen).

Is there any mechanism by which DKMS can be used?  Is there a change  that can be implemented?  Is it possible to put in place some sort of redirect for `uname -r` so that it will return the kernel version that pi-gen is working on?

Finally, other than the pi-gen issues is there some place more appropriate to raise the issue and to discuss this?

AaronDewes, 2020-12-24:
Can you share your changes in a public git repo so I can have a look at it?

Claudenw, 2020-12-24:
I am trying to install a couple of custom packages.  They are available at:

https://github.com/Claudenw/hd44780-i2c  <-- the HD44780-i2c driver
and 
https://github.com/Claudenw/hd44780-i2c-pkg <-- the DKMS  based packageing for the above driver

I suspect that the problems arise with any DKMS based packaging, however I will put together a public git repo based on pi-gen to show the problem.

AaronDewes, 2020-12-25:
Maybe this helps: https://github.com/AaronDewes/pi-gen/tree/fake-uname

You can run any command like this:

`fakeuname -s Linux theactualcommand`

You can pass any argument you want in the way uname would return it, fakeuname -s sets the output of uname -s, ...
You cant set -a, that's automatically generated by the other options.

AaronDewes, 2020-12-25:
You can find the uname strings here: https://github.com/raspberrypi/firmware/tree/stable/extra (`8` is for armv8/arm64, so  only the arm64 branch of this repo would have the string. `7l` is RPi4 and 400, `v7` is RPi 3 (and maybe 2, I'm not sure), the one without a suffix is for RPi1 and Zero

Claudenw, 2020-12-30:
Sorry it has taken so long to get back to you but work got in the way.  I have created a minimal example of the problem at https://github.com/Claudenw/pi-gen-485 

I am unable to verify that it fails as none of my docker based pi-gen compilations are working at the moment due to issues retrieving `adduser 3.118`.  I keep getting header resets.  I suspect that this is a transient issue and perhaps it is just local to me.  

The [readme](https://github.com/Claudenw/pi-gen-485/blob/main/README.md) explains how the system works but if you clone the repository and run `./setup.sh` it should set up a complete pi-gen build directory ready for building.

Claudenw, 2021-01-19:
@AaronDewes , From what I can see, I would need to be able to call fake-uname when I start the build or perhaps inside the build script.  but I do't have the kernel version number until the package is installed somewhere in step 0 or early step 1.  Do you see a way around this?

AaronDewes, 2021-01-19:
The fake-uname is always just added before the command you want to run, not just once.

Claudenw, 2021-01-19:
I am installing a debian package that uses DKMS in its install (in the postinst Debian script).  Since this is simply listed in an `nn-packages` file I would have to call fake-uname at [line 26 in the build.sh](https://github.com/RPi-Distro/pi-gen/blob/master/build.sh#L33), or I could call it where build.sh is called and have all of build.sh fall under it. (which might make sense).

Claudenw, 2021-01-20:
@AaronDewes , After doing some editing I added a call to `fakeuname` during the execution of `xx-packages` and `xx-packages-nr` files.  The text below is from the `xx-packages` processing:

```
					on_chroot << EOF
if [ -z "$KERNEL_VERSION" ]
then
        echo "Not calling fakeuname - KERNEL_VERSION=${KERNEL_VERSION}"
	apt-get -o APT::Acquire::Retries=3 install -y $PACKAGES
else
        echo "calling fakeuname -r ${KERNEL_VERSION} apt-get -o APT::Acquire::Retries=3 install -y \$PACKAGES"
	fakeuname -r ${KERNEL_VERSION} apt-get -o APT::Acquire::Retries=3 install -y $PACKAGES
fi
EOF

```
`KERNEL_VERSION` is set once the kernel headers have been installed.  The solution I came up with picks the common header version. (5.4.83+ in this case).  The resulting output line is:

```
calling fakeuname -r 5.4.83+ apt-get -o APT::Acquire::Retries=3 install -y 
```

However later during the installation of the DKMS dependent software I find this:

```
DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.0-62-generic cannot be found at
/lib/modules/5.4.0-62-generic/build or /lib/modules/5.4.0-62-generic/source.
dpkg: error processing package hd44780-i2c (--configure):
 installed hd44780-i2c package post-installation script subprocess returned error exit status 
```

`5.4.0-62-generic` is the version of the underlying OS on which pi-gen is running.  It looks like fakeuname is not getting passed down through the build chain.  I do not wantto modify the hd44780-i2c build as it should work in a normal environment as well as in the pi-gen build.  

A couple of possibilities present themselves to me, but I don't know enough about the pi-gen environment to understand which if any will work:

1. Modify fakeuname so that gets pushed down into the environments of the scripts called by the debian install process (e.g. preinst, postinst, prerm, postrm).
2. Create a symbolic link to fakename from /bin/uname and then undo that symbolic link at the end of processing.
3. Creae a "diversion" to fakename when KERNEL_VERSION has been set.  I have no idea how to do this.

Any assistance would be appreciated.

AaronDewes, 2021-01-20:
You need to install the raspberrypi-kernel-headers package in the chroot and modify the debian package somehow to pass the correct kernel version as -k to dkms. Not sure how to do that other than hardcoding it though

Claudenw, 2021-01-20:
DKMS selects the kernel version via a call the uname.  So if fakeuname can be pushed down to the debian script it should work.  However, that does not seem to be the case.

I have installed the raspberrypi-kernel-headers and used that to determine the kernel version.  Which is set in the KERNEL_VERSION variable and used to set the -r option on the fakeuname.  

the problem seems to be that I can insert the fake uname with
`fakeuname -r ${KERNEL_VERSION} -m arm6l -s Linux apt-get -o APT::Acquire::Retries=3 install --no-install-recommends -y $PACKAGES` and it will call `apt-get` and begin the installation.

However apt-get calls the debian script `postinst` that calls the `dkms` script.  The dkms script makes calls to uname.

so the stack is
```
fakeuname -r 5.4.83+ -m arm6l -s Linux
apt-get
postinst (shell script)
dkms (shell script)
uname -r
```

By the time the uname at the bottom gets called it does not return 5.4.83+ as expected.  Is there a way around this?  Is there any way to push the fakeuname that far down the stack?

Claudenw, 2021-02-15:
I found a solution.

As noted above the problem is to get the proper kernel version into the DKMS build.  I solved this by modifying pigen as well as the dkms call in the package being installed.

**PIGEN Changes**

in pigen/build.sh I added a function to capture the versions

```
check_kernel_version()
{
	if [ -z "${KERNEL_VERSION}" ]
	then
		if [ 0 -lt `find ${ROOTFS_DIR}/usr/src -name linux-headers\* -type d | wc -l` ]
		then
			for fn in `cat ${ROOTFS_DIR}/usr/src/linux-headers*/include/config/kernel.release`
			do
				export KERNEL_VERSION="${KERNEL_VERSION} -k ${fn}"
			done
		fi
		log "====> Kernel Version: '${KERNEL_VERSION}'"
	fi
}
```
 This function sets a `${KERNEL_VERSION}` environment variable if it is not set and the linux-headers have been installed.  It adds all the kernel versions to the `KERNEL_VERSION` variable as a series of `-k version` strings.

I modified the `run_sub_stage` function so that it calls `check_kernel_version` at the end of `${i}-packages-nr` and `${i}-packages` stages.  This picks up the versions as soon as the kernel headers are added.

**DKMS Project Changes**

in the `postinst` script for the dpgk installation of the DKMS based project, I added 

```
        if [ -n "${KERNEL_VERSION}" ]
        then
           DKMS_ARG="${KERNEL_VERSION}"
        fi
```
So that the `DKMS_ARG` contains the `-k <version>` entries from the pigen script.

I then modified the DKMS builds in the script

```
        dkms build ${DKMS_ARG} hd44780-i2c/1.0
        dkms autoinstall ${DKMS_ARG}  hd44780-i2c/1.0
```

This then tells dkms to use the versions specified with the `-k` option rather than asking the system what kernel it is building on.

XECDesign, 2021-06-22:
As noted in #493, I believe this is a packaging issue. dh_dkms correctly handles chroot environments. postinst scripts shouldn't not have manually added dkms commands like this.