RPi-Distro/pi-gen #487: Building image on btrfs

utement, 2021-01-08:
I can succesfully build btrfs root image, however I did got stuck on boot with kernel panic "Failed to execute /init: Kernel panic - not syncing: No init found". I did follow this tutorial as reference https://feldspaten.org/2019/07/05/raspbian-and-btrfs/.

I believe the issue is with comman mkinitramfs -o /boot/initramfs-btrfs.gz where I should probably specify pi-gen version, but I guess it takes default or vagrant box kernel version that I'm runing pi-gen build on.

How can I get pi-gen version in export-image scripts?

XECDesign, 2021-01-08:
fake-uname may help

Related - https://github.com/RPi-Distro/pi-gen/issues/485

utement, 2021-01-10:
I did create initramfs-btrfs.gz with standard pi-gen generated image, and copied it to my modified image. It does complain on load that parted, partition id is not correct, but with confirming it loads system. I still have to start working on resize script but before that I need to find out how can I build correct initramfs-btrfs.gz in pi-gen build time. 

Any idea how can I run mkinitramfs as it would be run on actual partition.

solskogen, 2023-11-09:
Now with bookworm I hope this can see some love :-)