RPi-Distro/pi-gen #495: [ARM64] Lock at Preconfiguring packages

zoic21, 2021-02-24:
Hello,

I have an issue since few month, build of arm64 freeze on ``Preconfiguring packages ...``. I test more than 24h and nothing more.

```
Sending build context to Docker daemon  810.5kB
Step 1/6 : ARG BASE_IMAGE=debian:buster
Step 2/6 : FROM ${BASE_IMAGE}
 ---> e44089caceeb
Step 3/6 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 50711c888205
Step 4/6 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         bsdtar libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates     && rm -rf /var/lib/apt/lists/*
 ---> [Warning] The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in 498b68e69b75
Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:2 http://deb.debian.org/debian buster InRelease [122 kB]
Get:3 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:4 http://security.debian.org/debian-security buster/updates/main i386 Packages [266 kB]
Get:5 http://deb.debian.org/debian buster/main i386 Packages [7864 kB]
Get:6 http://deb.debian.org/debian buster-updates/main i386 Packages [9512 B]
Fetched 8378 kB in 2s (3830 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
coreutils is already the newest version (8.30-3).
coreutils set to manually installed.
grep is already the newest version (3.3-1).
grep set to manually installed.
libcap2-bin is already the newest version (1:2.25-2).
libcap2-bin set to manually installed.
The following additional packages will be installed:
  bsdmainutils bzip2 diffstat dmidecode dmsetup gettext gettext-base git-man
  libarchive-tools libarchive13 libbsd0 libcroco3 libcurl3-gnutls libcurl4
  libdevmapper1.02.1 liberror-perl libexpat1 libgdbm-compat4 libgdbm6
  libglib2.0-0 libgomp1 libgpm2 libgssapi-krb5-2 libicu63 libk5crypto3
  libkeyutils1 libkmod2 libkrb5-3 libkrb5support0 libldap-2.4-2 libldap-common
  libmagic-mgc libmagic1 libncurses6 libnghttp2-14 libparted2 libpcre2-8-0
  libperl5.28 libpipeline1 libpopt0 libpsl5 libreadline7 librtmp1 libsasl2-2
  libsasl2-modules-db libssh2-1 libssl1.1 libxml2 lsb-base openssl patch perl
  perl-modules-5.28 readline-common sensible-utils vim-common vim-runtime wget
Suggested packages:
  cpp wamerican | wordlist whois vacation bzip2-doc ubuntu-archive-keyring
  squid-deb-proxy-client gettext-doc autopoint libasprintf-dev
  libgettextpo-dev git-daemon-run | git-daemon-sysvinit git-doc git-el
  git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn lrzip gdbm-l10n
  gpm krb5-doc krb5-user libparted-dev libparted-i18n parted-doc ed
  diffutils-doc perl-doc libterm-readline-gnu-perl
  | libterm-readline-perl-perl make libb-debug-perl liblocale-codes-perl sudo
  default-mta | mail-transport-agent graphviz procmail readline-doc
  openssh-client openssh-server ctags vim-doc vim-scripts
Recommended packages:
  gnupg arch-test less ssh-client libglib2.0-data shared-mime-info
  xdg-user-dirs krb5-locales publicsuffix libsasl2-modules netbase unzip
The following NEW packages will be installed:
  bc binfmt-support bsdmainutils bsdtar bzip2 ca-certificates curl debootstrap
  diffstat dmidecode dmsetup dosfstools file gettext gettext-base git git-man
  kmod libarchive-tools libarchive13 libbsd0 libcroco3 libcurl3-gnutls
  libcurl4 libdevmapper1.02.1 liberror-perl libexpat1 libgdbm-compat4 libgdbm6
  libglib2.0-0 libgomp1 libgpm2 libgssapi-krb5-2 libicu63 libk5crypto3
  libkeyutils1 libkmod2 libkrb5-3 libkrb5support0 libldap-2.4-2 libldap-common
  libmagic-mgc libmagic1 libncurses6 libnghttp2-14 libparted2 libpcre2-8-0
  libperl5.28 libpipeline1 libpopt0 libpsl5 libreadline7 librtmp1 libsasl2-2
  libsasl2-modules-db libssh2-1 libssl1.1 libxml2 lsb-base openssl parted
  patch perl perl-modules-5.28 qemu-user-static quilt readline-common rsync
  sensible-utils udev vim vim-common vim-runtime wget xxd xz-utils zerofree
  zip
0 upgraded, 78 newly installed, 0 to remove and 1 not upgraded.
Need to get 68.5 MB of archives.
[... 2555 lines cut ...]
Get:212 http://deb.debian.org/debian buster/main arm64 python3-six all 1.12.0-1 [15.7 kB]
Get:213 http://deb.debian.org/debian buster/main arm64 python3-urllib3 all 1.24.1-1 [97.1 kB]
Get:214 http://deb.debian.org/debian buster/main arm64 python3-requests all 2.21.0-1 [66.9 kB]
Get:215 http://deb.debian.org/debian buster/main arm64 usb.ids all 2019.07.27-0+deb10u1 [174 kB]
Get:216 http://deb.debian.org/debian buster/main arm64 usbutils arm64 1:010-3 [72.7 kB]
Get:217 http://deb.debian.org/debian buster/main arm64 rfkill arm64 2.33.1-0.1 [82.1 kB]
Get:218 http://deb.debian.org/debian buster/main arm64 rng-tools arm64 2-unofficial-mt.14-1+b2 [39.9 kB]
Get:219 http://deb.debian.org/debian buster/main arm64 rsync arm64 3.1.3-6 [385 kB]
Get:220 http://deb.debian.org/debian buster/main arm64 shared-mime-info arm64 1.10-1 [764 kB]
Get:221 http://deb.debian.org/debian buster/main arm64 ssh-import-id all 5.7-1 [19.8 kB]
Get:222 http://deb.debian.org/debian buster/main arm64 strace arm64 4.26-0.2 [803 kB]
Get:223 http://deb.debian.org/debian buster/main arm64 sudo arm64 1.8.27-1+deb10u3 [1208 kB]
Get:224 http://deb.debian.org/debian buster/main arm64 unzip arm64 6.0-23+deb10u2 [166 kB]
Get:225 http://deb.debian.org/debian buster/main arm64 libjim0.77 arm64 0.77+dfsg0-3 [110 kB]
Get:226 http://deb.debian.org/debian buster/main arm64 usb-modeswitch-data all 20170806-2 [45.5 kB]
Get:227 http://deb.debian.org/debian buster/main arm64 usb-modeswitch arm64 2.5.2+repack0-2 [55.9 kB]
Get:228 http://deb.debian.org/debian buster/main arm64 v4l-utils arm64 1.16.3-3 [453 kB]
Get:229 http://deb.debian.org/debian buster/main arm64 xauth arm64 1:1.0.10-1 [38.2 kB]
Get:230 http://deb.debian.org/debian buster/main arm64 xdg-user-dirs arm64 0.17-2 [53.2 kB]
Preconfiguring packages ...
```

I use docker build, did someone got these issue ?

Thank in advance.

mmmspatz, 2021-03-04:
It seems like grep hangs in the second stage of debootstrap. I tried running build.sh a debian buster i386 vm instead of using docker and hit the same issue.

Reverting dd96ca1c8629da6589b381546e2b7677c7611744 fixes it for me.

mmmspatz, 2021-03-04:
I think I must have assumed success too early, reverting that commit doesn't actually fix it. I'm still stuck on "Preconfiguring packages". The command that is hung is `/usr/bin/qemu-aarch64-static /usr/bin/grep -F _guess*`

zoic21, 2021-03-05:
No success for me also... It's stuck on same line

rob-clarke, 2021-03-13:
I'm seeing the same issue with a Docker build from Ubuntu 18 5.4.0-66-generic (unsupported).

That grep command appears to be from `/tmp/console-setup.config.{RANDOM}`. Specifically, it initially gets stuck in the call to `decode_codeset` within that script. If this is killed it then gets stuck in the `encode_codeset` function. Common to both of these is a preceding call to `echo "$codesets"`. Maybe `echo` isn't ending its output stream for some reason?

Regardless, removing `console-setup` from `stage2/01-sys-tweaks/00-packages` skips over this particular hang, but results in a segfault for me, YMMV. I'm not sure how critical that package is to the final system image.

mmmspatz, 2021-03-15:
Today I verified that the issue does not occur if you build on an aarch64 host.

I used my existing 64 bit raspios install on my rpi4 (~5 months old, I've never run apt upgrade on it) to successfully build a new image using the docker build. Incidentally, I had to revert dd96ca1c8629da6589b381546e2b7677c7611744 first, because it caused the i386 docker base image to be used even on a pi.

Then, I booted my new up-to-date image on my rpi3b and successfully built another image using the non-docker build and the current arm64 branch (041b97464c01ac6453e4aab59dc931220237bc16). I did it on a USB flash drive and it took literally all day, the 4 really is an incredible improvement.

So, I believe this issue has to do with using qemu in stage 2 of debootstrap.

XECDesign, 2021-09-20:
Haven't seen this issue myself. Does it happen if you don't use Docker at all? I've seen hangs similar to this when using quite old versions of qemu. I'd try building outside of docker or with a docker base image that's based on a recent OS release.