RPi-Distro/pi-gen #497: Possible to create a read-only-filesystem? Separate read-only (e.g. OS) and modifyable data (e.g. app data)?

capiman, 2021-03-15:
We use a Raspberry Pi in an environment, where we face the problem of sudden power loss.
We can't use an additional battery.
What would be best way to keep sure a sudden power loss does not kill our file system?

I am thinking about separating data into 2 areas: One in flash, where you can only read and NEVER write.
And one area (in RAM) where all temporary data is stored. A power loss would kill this data,
but on next startup you get a clean system and can start completely new.

Would this be possible with pi-gen? Has someone already done such a way?

utement, 2021-03-15:
I’m currently working same thing.

This is the answer https://github.com/JasperE84/root-ro

Uroš

XECDesign, 2021-09-20:
It's certainly possible. The approach I'd recommend involves using initramfs to set up an overlayfs. There's an option in raspi-config to do this, so you could port the change in that script across to a pi-gen stage.

However, it's a bit outside of the scope of this issue tracker. Maybe the forum would be a good place to get help with this.

https://www.raspberrypi.org/forums/

closed 2021-09-20 by XECDesign